eidottermihi/rpicheck #69: Banned for Google Play store?

poll0rz, 2015-02-12:
https://play.google.com/store/apps/details?id=de.eidottermihi.rpicheck

[image]

eidottermihi, 2015-02-12:
Will be back soon, was removed because of violation of Raspberry Pi Foundation registered trademark (logo)...

poll0rz, 2015-03-04:
still banned. any news please?
i liked the app and some friends want it.
thanks!

cbamofkd, 2015-03-04:
@eidottermihi
Can you please provide an alternative download until its back in the Play Store?
Publishing it in F-Droid would be even better!

eidottermihi, 2015-03-05:
Sorry guys, I can't get Google to reinstate the app, it's really frustrating. They refuse to reinstate the "old app" with changed logo etc. and also refuse to publish the "new app" (with new app package).

In the meantime I uploaded the [most recent version to Droidbin](http://www.droidbin.com/p19flrc15e1f071d8fs9d7ih1i8i3). Beware that the application has a new app package, so you have to add your Pi's again....

eidottermihi, 2015-03-10:
It's back online, but with different package name:

https://play.google.com/store/apps/details?id=de.eidottermihi.raspicheck

closed 2015-03-10 by eidottermihi