eidottermihi/rpicheck #84: Store logs in sdcard/Android/data instead sdcard/data

ogarcia, 2015-07-24:
The correct behaviour is that the app store our files into sdcard/Android/data instead use sdcard/data directly.

eidottermihi, 2015-07-24:
Why is sdcard/Android/data better in your opinion?

If we change this, you would need to implement some backward compability to open the old logs.

ogarcia, 2015-07-24:
Cause is the "Android Way":tm: :smile: and all other apps store their info in that folder. If in yor Adroid device you open sdcard/Android/data you can see a lot of folders with configs/logs/etc of several apps.

There is some references to this:
- In official docs: http://developer.android.com/reference/android/content/Context.html#getExternalFilesDir(java.lang.String)
- In github (different methods to obtain proper directory): https://gist.github.com/granoeste/5574148
- In stackexchange: http://android.stackexchange.com/questions/47924/where-android-apps-store-data

commit 5f946155 (2015-08-31): Move log to proper directory (fixes #84).
closed 2015-08-31 by eidottermihi