eidottermihi/rpicheck #85: Widget represent PI as offline

zoszkar, 2015-08-28:
I have Raspberry Pi 1 with latest OSMC 2015.06-1 (kernel version: 4.1.3-2-osmc) and my phone is an sammsung galaxy s4.

Problem is that the widget represent my pi as offline about half hour. After that if I open the app and close it, then the widget on next auto update (or manual update with refresh button) represents it again online. 

Here is the log:
============
19:06:11.744 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
19:06:11.948 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 22.
19:06:12.220 INFO  RaspiQuery - Using no host key verification.
19:06:57.047 INFO  RaspiQuery - Found vcgencmd in path: /opt/vc/bin/vcgencmd.
19:06:57.233 INFO  RaspiQuery - Querying core volts...
19:06:57.500 INFO  RaspiQuery - Querying memory information...
19:06:57.716 INFO  LoadAverageQuery - Querying load average for time period FIVE_MINUTES
19:06:57.814 INFO  RaspiQuery - Disconnecting from host ropi.no-ip.org.
19:11:12.304 DEBUG OverclockingWidget - Receiving Intent: action=updateOneWidget
19:11:12.312 DEBUG OverclockingWidget - Updating Widget[ID=117]. initByAlarm = true
19:11:12.348 DEBUG OverclockingWidget - Network available
19:11:12.349 DEBUG OverclockingWidget - Update was initiated by alarm.
19:11:12.349 DEBUG OverclockingWidget - Querying for Widget[ID=117]...
19:11:12.357 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
19:11:12.358 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 22.
19:11:12.394 INFO  RaspiQuery - Using no host key verification.
19:11:12.857 ERROR TransportImpl - Dying because - {}
net.schmizz.sshj.common.SSHRuntimeException: null
	at net.schmizz.sshj.common.Buffer.readPublicKey(Buffer.java:432) ~[na:0.0]
	at net.schmizz.sshj.transport.kex.AbstractDHG.next(AbstractDHG.java:108) ~[na:0.0]
	at net.schmizz.sshj.transport.KeyExchanger.handle(KeyExchanger.java:352) ~[na:0.0]
	at net.schmizz.sshj.transport.TransportImpl.handle(TransportImpl.java:484) ~[na:0.0]
	at net.schmizz.sshj.transport.Decoder.decode(Decoder.java:107) ~[na:0.0]
	at net.schmizz.sshj.transport.Decoder.received(Decoder.java:175) ~[na:0.0]
	at net.schmizz.sshj.transport.Reader.run(Reader.java:61) ~[na:0.0]
Caused by: java.security.GeneralSecurityException: java.security.NoSuchAlgorithmException: KeyFactory ECDSA implementation not found
	at net.schmizz.sshj.common.KeyType$3.readPubKeyFromBuffer(KeyType.java:146) ~[na:0.0]
	at net.schmizz.sshj.common.Buffer.readPublicKey(Buffer.java:430) ~[na:0.0]
	... 6 common frames omitted
Caused by: java.security.NoSuchAlgorithmException: KeyFactory ECDSA implementation not found
	at org.apache.harmony.security.fortress.Engine.notFound(Engine.java:190) ~[na:0.0]
	at org.apache.harmony.security.fortress.Engine.getInstance(Engine.java:139) ~[na:0.0]
	at java.security.KeyFactory.getInstance(KeyFactory.java:81) ~[na:0.0]
	at net.schmizz.sshj.common.KeyType$3.readPubKeyFromBuffer(KeyType.java:143) ~[na:0.0]
	... 7 common frames omitted
19:11:12.865 ERROR Promise - <<kex done>> woke to: net.schmizz.sshj.transport.TransportException
19:11:12.871 DEBUG OverclockingWidget - Query failed, showing device as offline.
19:11:12.871 DEBUG OverclockingWidget - Updating widget[ID=117] view.

eidottermihi, 2015-08-30:
This seems to be the case when Spongycastle wasn't registered as JCE provider before the update query is started. I will look into this...

zoszkar, 2015-08-31:
Thanks. If you have any question related to the bug, please ask me..

commit 6a248997 (2015-08-31): Move registering Spongycastle JCE provider to actual query class (should fix #85).
closed 2015-08-31 by eidottermihi