eidottermihi/rpicheck #86: No overclock information when running Ubuntu mate.

wingchun1274, 2015-08-29:
Hello!

I installed Ubuntu mate on my Raspberry pi 2 and I am wondering if you can support overclock information for Ubuntu mate. Other information is displayed correctly. Thanks in advance

eidottermihi, 2015-08-29:
Please post the debug log (Settings -> Show Log) here.

wingchun1274, 2015-08-30:
Screenshot and log :)

[image]

19:51:18.304 DEBUG MainActivity - Device cursor rows: 1
19:51:18.529 DEBUG MainActivity - Spinner item selected: pos=0, id=1
19:51:20.437 DEBUG MainActivity - Load average preference: ONE_MINUTE
19:51:20.443 INFO  RaspiQuery - New RaspiQuery for host: 192.168.1.110
19:51:20.445 INFO  RaspiQuery - Connecting to host: 192.168.1.110 on port 22.
19:51:20.569 INFO  RaspiQuery - Using no host key verification.
19:51:21.945 DEBUG RaspiQuery - Querying vcgencmd...
19:51:22.699 DEBUG RaspiQuery - Checking vcgencmd location: vcgencmd
19:51:22.790 DEBUG RaspiQuery - Path check output: VCHI initialization failed

19:51:22.791 INFO  RaspiQuery - Found vcgencmd in path: vcgencmd.
19:51:22.856 ERROR RaspiQuery - Could not parse frequency.
19:51:22.857 ERROR RaspiQuery - Output of 'vcgencmd measure_clock [core/arm]': 
VCHI initialization failed

19:51:22.921 ERROR RaspiQuery - Could not parse frequency.
19:51:22.922 ERROR RaspiQuery - Output of 'vcgencmd measure_clock [core/arm]': 
VCHI initialization failed

19:51:22.923 INFO  RaspiQuery - Querying core volts...
19:51:22.992 ERROR RaspiQuery - Could not parse cpu voltage.
19:51:22.993 ERROR RaspiQuery - Output of 'vcgencmd measure_volts core': 
VCHI initialization failed

19:51:23.061 ERROR RaspiQuery - Could not parse cpu temperature.
19:51:23.066 ERROR RaspiQuery - Output of 'vcgencmd measure_temp': 
VCHI initialization failed

19:51:23.069 DEBUG RaspiQuery - Querying firmware version...
19:51:23.138 ERROR RaspiQuery - Could not parse firmware. Maybe the output of 'vcgencmd version' changed.
19:51:23.138 DEBUG RaspiQuery - Output of 'vcgencmd version': 
VCHI initialization failed

19:51:23.139 DEBUG RaspiQuery - Firmware version: n/a
19:51:23.141 INFO  LoadAverageQuery - Querying load average for time period ONE_MINUTE
19:51:23.309 DEBUG LoadAverageQuery - Checking line: 0.11 0.06 0.21 1/297 1759
19:51:23.310 DEBUG LoadAverageQuery - Checking line: 5
19:51:23.320 INFO  UptimeQuery - Querying uptime...
19:51:23.392 INFO  RaspiQuery - Querying memory information...
19:51:23.469 INFO  SerialNoQuery - Querying serial number...
19:51:23.540 INFO  RaspiQuery - Querying running processes...
19:51:23.708 INFO  RaspiQuery - Querying network interfaces...
19:51:23.754 DEBUG RaspiQuery - Found interface eth0.
19:51:23.755 INFO  RaspiQuery - Available interfaces: [eth0]
19:51:23.756 INFO  RaspiQuery - Checking carrier of eth0...
19:51:23.821 DEBUG RaspiQuery - eth0 has a carrier up and running.
19:51:23.822 INFO  RaspiQuery - Querying ip address of interface eth0...
19:51:23.988 DEBUG RaspiQuery - Output of ip query:     inet 192.168.1.110/24 brd 192.168.1.255 scope global dynamic eth0
.
19:51:23.989 INFO  RaspiQuery - eth0 - IP address: 192.168.1.110.
19:51:23.991 INFO  RaspiQuery - Querying disk usage...
19:51:24.060 INFO  RaspiQuery - Querying distribution name...
19:51:24.133 INFO  RaspiQuery - Disconnecting from host 192.168.1.110.
19:51:24.192 DEBUG SSHQueryTask - Query time: 3747 ms.
19:51:28.073 DEBUG MainActivity - Saving instance state (current device)
19:51:28.074 DEBUG MainActivity - Adding current device to all devices.

eidottermihi, 2015-08-30:
It looks just the same as #41.
Try to add your ssh user to group 'video', as described here http://raspberrypi.stackexchange.com/questions/7546/munin-node-plugins-vchi-initialization-failed/8423#8423 or here https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=34076

wingchun1274, 2015-08-31:
Thank you for responding that fast.
Running the command "usermod -a -G video myuser" worked like a charm.

closed 2015-08-31 by wingchun1274