eidottermihi/rpicheck #96: Release build not working

eidottermihi, 2015-11-05:
Stacktrace:

<pre><code>
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck E/RClassHelper﹕ No R class found for given package name de.eidottermihi.raspicheck
    java.lang.ClassNotFoundException: de.eidottermihi.raspicheck.R
            at java.lang.Class.classForName(Native Method)
            at java.lang.Class.forName(Class.java:309)
            at java.lang.Class.forName(Class.java:273)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromPackageName(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromApplication(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromActivity(Unknown Source)
            at de.fhconfig.android.library.injection.xml.ActivityXmlInjector.<init>(Unknown Source)
            at de.fhconfig.android.library.ui.injection.InjectionAppCompatActivity.onCreate(Unknown Source)
            at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(Unknown Source)
            at android.app.Activity.performCreate(Activity.java:6034)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2293)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2402)
            at android.app.ActivityThread.access$900(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1318)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5273)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "de.eidottermihi.raspicheck.R" on path: DexPathList[[zip file "/data/app/de.eidottermihi.raspicheck-2/base.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
            at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
            at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
            at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
            at java.lang.Class.classForName(Native Method)
            at java.lang.Class.forName(Class.java:309)
            at java.lang.Class.forName(Class.java:273)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromPackageName(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromApplication(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromActivity(Unknown Source)
            at de.fhconfig.android.library.injection.xml.ActivityXmlInjector.<init>(Unknown Source)
            at de.fhconfig.android.library.ui.injection.InjectionAppCompatActivity.onCreate(Unknown Source)
            at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(Unknown Source)
            at android.app.Activity.performCreate(Activity.java:6034)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2293)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2402)
            at android.app.ActivityThread.access$900(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1318)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5273)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
    Suppressed: java.lang.ClassNotFoundException: de.eidottermihi.raspicheck.R
            at java.lang.Class.classForName(Native Method)
            at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
            at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
            at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
            ... 23 more
     Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack available
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ java.lang.ClassNotFoundException: de.eidottermihi.raspicheck.R
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at java.lang.Class.classForName(Native Method)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at java.lang.Class.forName(Class.java:309)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at java.lang.Class.forName(Class.java:273)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromPackageName(Unknown Source)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromApplication(Unknown Source)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromActivity(Unknown Source)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at de.fhconfig.android.library.injection.xml.ActivityXmlInjector.<init>(Unknown Source)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at de.fhconfig.android.library.ui.injection.InjectionAppCompatActivity.onCreate(Unknown Source)
11-05 22:25:15.648  14649-14649/de.eidottermihi.raspicheck W/System.err﹕ at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(Unknown Source)
[... 34 lines cut ...]
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5273)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
     Caused by: de.fhconfig.android.library.injection.exceptions.InjectionException: Could not determine which Annotation to inject.
            at de.fhconfig.android.library.injection.xml.XmlInjector.injectAnnotations(Unknown Source)
            at de.fhconfig.android.library.ui.injection.InjectionAppCompatActivity.onCreate(Unknown Source)
            at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(Unknown Source)
            at android.app.Activity.performCreate(Activity.java:6034)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2293)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2402)
            at android.app.ActivityThread.access$900(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1318)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5273)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
11-05 22:25:18.074  14649-14649/de.eidottermihi.raspicheck I/Process﹕ Sending signal. PID: 14649 SIG: 9
</code></pre>

eidottermihi, 2015-11-05:
@larsgrefer Do we have to add a ProGuard rule for your libary? Without Proguard in debug build evertything is working fine.

larsgrefer, 2015-11-05:
Let me take a look at it

closed 2015-11-06 by eidottermihi

eidottermihi, 2015-11-06:
Build variant 'release' is still not working. Logcat:
<pre><code>
11-06 17:29:05.214  28766-28766/de.eidottermihi.raspicheck E/RClassHelper﹕ No R class found for given package name de.eidottermihi.raspicheck
    java.lang.ClassNotFoundException: de.eidottermihi.raspicheck.R
            at java.lang.Class.classForName(Native Method)
            at java.lang.Class.forName(Class.java:309)
            at java.lang.Class.forName(Class.java:273)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromPackageName(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromApplication(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromActivity(Unknown Source)
            at de.fhconfig.android.library.injection.xml.ActivityXmlInjector.<init>(Unknown Source)
            at de.fhconfig.android.library.ui.injection.InjectionAppCompatActivity.onCreate(Unknown Source)
            at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(Unknown Source)
            at android.app.Activity.performCreate(Activity.java:6034)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2293)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2402)
            at android.app.ActivityThread.access$900(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1318)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5273)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "de.eidottermihi.raspicheck.R" on path: DexPathList[[zip file "/data/app/de.eidottermihi.raspicheck-1/base.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]
            at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
            at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
            at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
            at java.lang.Class.classForName(Native Method)
            at java.lang.Class.forName(Class.java:309)
            at java.lang.Class.forName(Class.java:273)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromPackageName(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromApplication(Unknown Source)
            at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromActivity(Unknown Source)
            at de.fhconfig.android.library.injection.xml.ActivityXmlInjector.<init>(Unknown Source)
            at de.fhconfig.android.library.ui.injection.InjectionAppCompatActivity.onCreate(Unknown Source)
            at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(Unknown Source)
            at android.app.Activity.performCreate(Activity.java:6034)
            at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
            at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2293)
            at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2402)
            at android.app.ActivityThread.access$900(ActivityThread.java:151)
            at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1318)
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5273)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
    Suppressed: java.lang.ClassNotFoundException: de.eidottermihi.raspicheck.R
            at java.lang.Class.classForName(Native Method)
            at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
            at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
            at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
            ... 23 more
     Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack available
11-06 17:29:05.214  28766-28766/de.eidottermihi.raspicheck W/System.err﹕ java.lang.ClassNotFoundException: de.eidottermihi.raspicheck.R
11-06 17:29:05.214  28766-28766/de.eidottermihi.raspicheck W/System.err﹕ at java.lang.Class.classForName(Native Method)
11-06 17:29:05.214  28766-28766/de.eidottermihi.raspicheck W/System.err﹕ at java.lang.Class.forName(Class.java:309)
11-06 17:29:05.214  28766-28766/de.eidottermihi.raspicheck W/System.err﹕ at java.lang.Class.forName(Class.java:273)
11-06 17:29:05.214  28766-28766/de.eidottermihi.raspicheck W/System.err﹕ at de.fhconfig.android.library.injection.helper.RClassHelper.getRClassFromPackageName(Unknown Source)
[... cut ...]
            at android.os.Handler.dispatchMessage(Handler.java:102)
            at android.os.Looper.loop(Looper.java:135)
            at android.app.ActivityThread.main(ActivityThread.java:5273)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:700)
</code></pre>

eidottermihi, 2015-11-06:
Nevermind, after a 'clean' everything works.

PR #97 (2015-11-06): Update the android library
fixes #96