eidottermihi/rpicheck #100: Widget stops refreshing after few hours

zoszkar, 2015-11-13:
I have Raspberry Pi 1 with latest OSMC 2015.09-1 (kernel version: 4.2.1-1-osmc) and my phone is an samsung galaxy s4.
Widget stops refreshing if the phone is long time in sleep mode. I see in the log ```TransportException: Broken transport; encountered EOF``` exception. The AlarmManager doesn't stop always immediately, sometimes the exception repeats periodically 2-3 times.
If the AlarmManager sopped working (onReceive function doesn't called  anymore) the widget stay in refresh mode so always showing the progress bar.
If I open and close the app after the AlarmManager has been stopped, it works again.

Here is the log:
```
00:11:18.259 DEBUG OverclockingWidget - Receiving Intent: action=updateOneWidget
00:11:18.261 DEBUG OverclockingWidget - Updating Widget[ID=178]. initByAlarm = true
00:11:18.275 DEBUG OverclockingWidget - Network available
00:11:18.275 DEBUG OverclockingWidget - Update was initiated by alarm.
00:11:18.276 DEBUG OverclockingWidget - Querying for Widget[ID=178]...
00:11:18.279 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
00:11:18.279 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 56565.
00:11:18.319 INFO  RaspiQuery - Using no host key verification.
00:12:42.631 DEBUG RaspiQuery - Querying vcgencmd...
00:12:42.976 DEBUG RaspiQuery - Checking vcgencmd location: vcgencmd
00:12:43.006 DEBUG RaspiQuery - Path check output: bash: vcgencmd: command not found

00:12:43.069 DEBUG RaspiQuery - Checking vcgencmd location: /usr/bin/vcgencmd
00:12:43.141 DEBUG RaspiQuery - Path check output: bash: /usr/bin/vcgencmd: No such file or directory

00:12:43.189 DEBUG RaspiQuery - Checking vcgencmd location: /opt/vc/bin/vcgencmd
00:12:43.229 DEBUG RaspiQuery - Path check output: 
00:12:43.230 INFO  RaspiQuery - Found vcgencmd in path: /opt/vc/bin/vcgencmd.
00:12:43.398 INFO  RaspiQuery - Querying core volts...
00:12:43.576 DEBUG RaspiQuery - Querying firmware version...
00:12:43.679 DEBUG RaspiQuery - Firmware version: c156d00b148c30a3ba28ec376c9c01e95a77d6d5 (clean) (release)
00:12:43.681 INFO  RaspiQuery - Querying memory information...
00:13:10.319 INFO  LoadAverageQuery - Querying load average for time period FIVE_MINUTES
00:13:10.413 DEBUG LoadAverageQuery - Checking line: 0.07 0.10 0.12 1/122 2641
00:13:10.414 DEBUG LoadAverageQuery - Checking line: 2
00:13:10.415 INFO  RaspiQuery - Disconnecting from host ropi.no-ip.org.
00:13:10.417 DEBUG OverclockingWidget - min: 0.0  max: 90.0 value: 43.3
00:13:10.417 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 48.11111111111111
00:13:10.418 DEBUG OverclockingWidget - min: 500.0  max: 1200.0 value: 1000.0
00:13:10.418 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 71.42857142857143
00:13:10.419 DEBUG OverclockingWidget - min: 0.0  max: 100.0 value: 10.0
00:13:10.419 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 10.0
00:13:10.419 DEBUG OverclockingWidget - min: 0.0  max: 100.0 value: 83.0
00:13:10.420 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 83.0
00:13:10.420 DEBUG OverclockingWidget - Updating widget[ID=178] view.
00:16:19.569 DEBUG OverclockingWidget - Receiving Intent: action=updateOneWidget
00:16:19.577 DEBUG OverclockingWidget - Updating Widget[ID=178]. initByAlarm = true
00:16:19.589 DEBUG OverclockingWidget - Network available
00:16:19.590 DEBUG OverclockingWidget - Update was initiated by alarm.
00:16:19.590 DEBUG OverclockingWidget - Querying for Widget[ID=178]...
00:16:19.592 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
00:16:19.593 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 56565.
00:16:19.632 INFO  RaspiQuery - Using no host key verification.
00:21:01.499 ERROR TransportImpl - Dying because - {}
net.schmizz.sshj.transport.TransportException: Broken transport; encountered EOF
	at net.schmizz.sshj.transport.Reader.run(Reader.java:59) ~[na:0.0]
00:21:19.294 DEBUG OverclockingWidget - Receiving Intent: action=updateOneWidget
00:21:19.300 DEBUG OverclockingWidget - Updating Widget[ID=178]. initByAlarm = true
00:21:19.346 DEBUG OverclockingWidget - Network available
00:21:19.346 DEBUG OverclockingWidget - Update was initiated by alarm.
00:21:19.347 DEBUG OverclockingWidget - Querying for Widget[ID=178]...
00:21:19.588 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
00:21:19.590 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 56565.
00:21:19.833 INFO  RaspiQuery - Using no host key verification.
00:23:11.379 DEBUG RaspiQuery - Querying vcgencmd...
00:23:19.334 DEBUG RaspiQuery - Checking vcgencmd location: vcgencmd
00:23:19.374 DEBUG RaspiQuery - Path check output: bash: vcgencmd: command not found

00:23:19.417 DEBUG RaspiQuery - Checking vcgencmd location: /usr/bin/vcgencmd
00:23:19.451 DEBUG RaspiQuery - Path check output: bash: /usr/bin/vcgencmd: No such file or directory

00:23:19.497 DEBUG RaspiQuery - Checking vcgencmd location: /opt/vc/bin/vcgencmd
00:23:19.543 DEBUG RaspiQuery - Path check output: 
00:23:19.544 INFO  RaspiQuery - Found vcgencmd in path: /opt/vc/bin/vcgencmd.
00:23:19.747 INFO  RaspiQuery - Querying core volts...
00:23:19.913 DEBUG RaspiQuery - Querying firmware version...
00:23:20.004 DEBUG RaspiQuery - Firmware version: c156d00b148c30a3ba28ec376c9c01e95a77d6d5 (clean) (release)
00:23:20.046 INFO  RaspiQuery - Querying memory information...
00:25:51.265 INFO  LoadAverageQuery - Querying load average for time period FIVE_MINUTES
00:25:51.368 DEBUG LoadAverageQuery - Checking line: 0.10 0.10 0.12 1/122 2714
00:25:51.370 DEBUG LoadAverageQuery - Checking line: 2
00:25:51.371 INFO  RaspiQuery - Disconnecting from host ropi.no-ip.org.
00:25:51.382 DEBUG OverclockingWidget - min: 0.0  max: 90.0 value: 43.3
00:25:51.383 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 48.11111111111111
00:25:51.385 DEBUG OverclockingWidget - min: 500.0  max: 1200.0 value: 1000.0
00:25:51.386 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 71.42857142857143
00:25:51.387 DEBUG OverclockingWidget - min: 0.0  max: 100.0 value: 10.0
00:25:51.388 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 10.0
00:25:51.390 DEBUG OverclockingWidget - min: 0.0  max: 100.0 value: 84.0
00:25:51.391 DEBUG OverclockingWidget - Updating progressbar: scaledValue = 84.0
00:25:51.392 DEBUG OverclockingWidget - Updating widget[ID=178] view.
00:26:18.281 DEBUG OverclockingWidget - Receiving Intent: action=updateOneWidget
00:26:18.291 DEBUG OverclockingWidget - Updating Widget[ID=178]. initByAlarm = true
00:26:18.362 DEBUG OverclockingWidget - Network available
00:26:18.367 DEBUG OverclockingWidget - Update was initiated by alarm.
00:26:18.368 DEBUG OverclockingWidget - Querying for Widget[ID=178]...
00:26:18.377 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
00:26:18.378 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 56565.
00:26:18.450 INFO  RaspiQuery - Using no host key verification.
00:31:18.249 DEBUG OverclockingWidget - Receiving Intent: action=updateOneWidget
00:31:18.250 DEBUG OverclockingWidget - Updating Widget[ID=178]. initByAlarm = true
00:31:18.375 ERROR TransportImpl - Dying because - {}
net.schmizz.sshj.transport.TransportException: Broken transport; encountered EOF
	at net.schmizz.sshj.transport.Reader.run(Reader.java:59) ~[na:0.0]
00:31:18.384 DEBUG OverclockingWidget - Network available
00:31:18.385 DEBUG OverclockingWidget - Update was initiated by alarm.
00:31:18.385 DEBUG OverclockingWidget - Querying for Widget[ID=178]...
00:31:18.388 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
00:31:18.390 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 56565.
13:16:18.126 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
13:16:18.134 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 56565.
13:16:18.238 INFO  RaspiQuery - Using no host key verification.
13:16:22.726 INFO  RaspiQuery - Found vcgencmd in path: /opt/vc/bin/vcgencmd.
13:16:23.317 INFO  RaspiQuery - Querying core volts...
13:16:23.772 INFO  RaspiQuery - Querying memory information...
13:16:24.267 INFO  LoadAverageQuery - Querying load average for time period FIVE_MINUTES
13:16:24.524 INFO  RaspiQuery - Disconnecting from host ropi.no-ip.org.
13:21:17.988 INFO  RaspiQuery - New RaspiQuery for host: ropi.no-ip.org
13:21:17.990 INFO  RaspiQuery - Connecting to host: ropi.no-ip.org on port 56565.
13:21:18.043 INFO  RaspiQuery - Using no host key verification.
13:21:22.981 INFO  RaspiQuery - Found vcgencmd in path: /opt/vc/bin/vcgencmd.
13:21:23.455 INFO  RaspiQuery - Querying core volts...
13:21:24.113 INFO  RaspiQuery - Querying memory information...
13:21:25.124 INFO  LoadAverageQuery - Querying load average for time period FIVE_MINUTES
13:21:25.740 INFO  RaspiQuery - Disconnecting from host ropi.no-ip.org.
```
Before 13:16 opened and closed I the app, so you can see that the widget work again but without debug info...

eidottermihi, 2015-11-14:
Could you dump your alarm after widget update is'nt working any more with 
<code>
adb shell dumpsys alarm > dump.txt
</code>
and post it here?

eidottermihi, 2015-11-14:
Hm, it seems like that the AsyncTask, in which the actual query runs, will be killed by Android when device is in sleep and onReceive() of the alarm is finished.

Android documentation suggests to use a Service for this, so thats something that definitely should be changed.

zoszkar, 2015-11-16:
Probably that’s the problem...
When I attach the debugger to the emulator or phone the bug doesn't appear.

indywidualny, 2016-01-28:
Well. I have a similar problem. Widget is not being automatically refreshed at all. Only manual refresh works. After adding it I have to force sync it in order to notice any data. And to update it I have to force sync and so on...

zoszkar, 2016-01-29:
I'm not sure, but I think this bug is related to something else..

eidottermihi, 2016-01-30:
@indywidualny i think #117 is what you described?

indywidualny, 2016-01-30:
Yes but in fact it never updates. When I'm adding it there is no data until I refresh it manually. Later I have to refresh it manually too in order to update the widget. It's not updated automatically.

eidottermihi, 2016-02-02:
I already fixed that, but it's not released yet.
I want to fix this issue too before next release.

commit 3ff6848c (2016-02-03): Merge pull request #118 from eidottermihi/issue100

fix #100
closed 2016-02-03 by eidottermihi