eidottermihi/rpicheck #109: Overclocking section empty (Arch Linux)

indywidualny, 2016-01-26:
Zeros everywhere. I will try to fix it. 

```
12:00:27.499 DEBUG RaspiQuery - Querying vcgencmd...
12:00:28.094 DEBUG RaspiQuery - Checking vcgencmd location: vcgencmd
12:00:28.163 DEBUG RaspiQuery - Path check output: zsh:1: command not found: vcgencmd

12:00:28.349 DEBUG RaspiQuery - Checking vcgencmd location: /usr/bin/vcgencmd
12:00:28.430 DEBUG RaspiQuery - Path check output: zsh:1: no such file or directory: /usr/bin/vcgencmd

12:00:28.431 INFO  RaspiQuery - Found vcgencmd in path: /usr/bin/vcgencmd.
12:00:28.642 ERROR RaspiQuery - Could not parse frequency.
12:00:28.643 ERROR RaspiQuery - Output of 'vcgencmd measure_clock [core/arm]': 

12:00:28.897 ERROR RaspiQuery - Could not parse frequency.
12:00:28.898 ERROR RaspiQuery - Output of 'vcgencmd measure_clock [core/arm]': 

12:00:28.899 INFO  RaspiQuery - Querying core volts...
12:00:29.126 ERROR RaspiQuery - Could not parse cpu voltage.
12:00:29.128 ERROR RaspiQuery - Output of 'vcgencmd measure_volts core': 

12:00:29.418 ERROR RaspiQuery - Could not parse cpu temperature.
12:00:29.424 ERROR RaspiQuery - Output of 'vcgencmd measure_temp': 

12:00:29.426 DEBUG RaspiQuery - Querying firmware version...
12:00:29.598 ERROR RaspiQuery - Could not parse firmware. Maybe the output of 'vcgencmd version' changed.
12:00:29.599 DEBUG RaspiQuery - Output of 'vcgencmd version': 

12:00:29.600 DEBUG RaspiQuery - Firmware version: n/a
```

eidottermihi, 2016-01-26:
Looks like #41 #86 to me.

indywidualny, 2016-01-26:
Nope. It's not the issue. I'm the member of video.

https://github.com/phortx/Raspberry-Pi-Setup-Guide/blob/master/README.md
Section 4.5

Something is wrong with the path to vcgencmd. I added it to /etc/profile 

The app cannot find it though.

eidottermihi, 2016-01-26:
12:00:28.431 INFO  RaspiQuery - Found vcgencmd in path: /usr/bin/vcgencmd.

Can you execute /usr/bin/vcgencmd?

indywidualny, 2016-01-26:
```
pizero $ ~  LC_ALL=C /usr/bin/vcgencmd
zsh: no such file or directory: /usr/bin/vcgencmd
```
;)

indywidualny, 2016-01-26:
But the app can find the command because it's in my /etc/profile

Please give me your output of vcgencmd. I will make a simple workaround for my machine, creating a script that will format the data to be readable for the app. 

I will place it in /usr/bin so any changes will not be needed inside the app.

closed 2016-01-26 by eidottermihi

eidottermihi, 2016-01-26:
I found the problem in the code and fixed it, thanks!

Meanwhile, you can create a symbolic link in /usr/bin which points to /opt/vc/bin:
```
ln -s /opt/vc/bin/vcgencmd /usr/bin
```

indywidualny, 2016-01-26:
Awesome!

commit 226cd8a4 (2016-01-26): Rely more on exit status when searching vcgencmd location. Should fix #109