eidottermihi/rpicheck #110: Lack of Runtime Permissions support (Marshmallow)

indywidualny, 2016-01-26:
In Marshmallow,  it's not possible to save logs to file without granting WRITE_EXTERNAL_STORAGE at runtime. In order to save logs I had to grant permissions for this app manually (settings -> apps -> this app). 

This permission should be requested during an activation of "save logs".

The second (Phone) permission should be requested as well,  when the app is trying to use it.

Obviously before using a "dangerous permission" we need to make sure it's granted. If it's not an app should request it and the current action should be continued / stopped in `onRequestPermissionsResult`

eidottermihi, 2016-01-26:
We don't need permission WRITE_EXTERNAL_STORAGE at all on API > 18 if we store the log file according to [Android docs](http://developer.android.com/reference/android/content/Context.html#getExternalFilesDir%28java.lang.String%29).

I'm curious about the Phone permission, because the app doesnt need it and doesnt declare it in the AndroidManifest.xml...
I'll have to look into that.

indywidualny, 2016-01-26:
I'm just saying that my logs are saved in here:
/storage/emulated/0/Android/app_package/

Without granting storage permission by hand logs are not saved. I'm talking about API 23, latest Marshmallow (CM 13). I'm not sure why this happens at my Nexus 4, but it happens. Maybe Cyanogenmod Team did something wrong, maybe Google Team... ;)

You are right. According to the documentation there shouldn't be any problem. Everything is OK, in theory. 

But it's only my device and nobody but me is complaining so it's not really the issue. I was able to grant the missing permission by hand in order to save logs ;) 

About 'Phone' permission. It's actually something from Phone group. I'm not sure what exactly but I had a similar bug in my app in the past. Something was automatically adding a permission during a signed app export. The solution was strange. I had to target one of my projects dependencies to the same API as the app.

larsgrefer, 2016-01-26:
@indywidualny I'm afraid this is my fault. See freefair/android-colors#1

indywidualny, 2016-01-26:
So just like I thought. A dependency :) Great.

eidottermihi, 2016-01-27:
Phone should be gone after https://github.com/freefair/android-util/pull/1 is released.

eidottermihi, 2016-01-28:
Waiting on https://github.com/freefair/android-library/pull/22 to get merged and released

PR #111 (2016-01-26): Update dependency version
fix #110 

see also: freefair/android-colors#1
PR #116 (2016-01-29): Update dependencies
fix #110
see freefair/android-library#22
closed 2016-01-29 by eidottermihi