eidottermihi/rpicheck #121: App crashes after connect attempt

Offerel, 2016-03-29:
Since some days the App crashes after I try to connect to my Raspberry pi. I get the Message "App angehalten", since it's a german Android Phone (Samsung Galaxy S6 with Android 6). I connect via username and password. I don't know if this is a Marshmallow Issue. 

I have no problem in the log:

22:39:31.853 [AsyncTask #7] INFO  d.e.r.ssh.impl.queries.MemoryQuery - Querying memory information...
22:39:31.933 [AsyncTask #7] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Disconnecting from host 192.168.1.12.
22:39:58.299 [main] WARN  d.e.r.activity.SettingsActivity - Enabling debug logging. Be warned that the log file can get huge because of this.
22:39:58.305 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
22:40:02.425 [main] DEBUG d.e.rpicheck.activity.MainActivity - Load average preference: FIVE_MINUTES
22:40:02.479 [AsyncTask #4] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - New RaspiQuery for host: 192.168.1.12
22:40:02.479 [AsyncTask #4] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host: 192.168.1.12 on port 22.
22:40:02.509 [AsyncTask #4] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification.
22:40:03.260 [AsyncTask #4] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Querying vcgencmd...
22:40:03.276 [AsyncTask #4] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Checking vcgencmd location: vcgencmd
22:40:03.327 [AsyncTask #4] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Path check output:  22:40:04 up 1 day,  5:04,  1 user,  load average: 0,09, 0,07, 0,05

22:40:03.328 [AsyncTask #4] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Found vcgencmd in path: vcgencmd.
22:40:03.505 [AsyncTask #4] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Querying core volts...
22:40:03.686 [AsyncTask #4] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Querying firmware version, vcgencmd path=vcgencmd
22:40:03.775 [AsyncTask #4] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Firmware version: Copyright (c) 2012 Broadcom
22:40:03.776 [AsyncTask #4] INFO  d.e.r.s.i.queries.LoadAverageQuery - Querying load average for time period FIVE_MINUTES
22:40:03.898 [AsyncTask #4] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line:  22:40:04 up 1 day,  5:04,  1 user,  load average: 0,09, 0,07, 0,05
22:40:03.899 [AsyncTask #4] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line:  22:40:04 up 1 day,  5:04,  1 user,  load average: 0,09, 0,07, 0,05
22:40:03.900 [AsyncTask #4] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: 0.09 0.07 0.05 1/147 24739
22:40:03.901 [AsyncTask #4] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: 5
22:40:03.903 [AsyncTask #4] INFO  d.e.r.ssh.impl.queries.UptimeQuery - Querying uptime...
22:40:04.003 [AsyncTask #4] DEBUG d.e.r.ssh.impl.queries.UptimeQuery - Skipping line:  22:40:04 up 1 day,  5:04,  0 users,  load average: 0,09, 0,07, 0,05
22:40:04.005 [AsyncTask #4] INFO  d.e.r.ssh.impl.queries.MemoryQuery - Querying memory information...
22:40:04.084 [AsyncTask #4] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Disconnecting from host 192.168.1.12.
22:40:13.650 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
22:40:13.668 [AsyncTask #2] DEBUG d.e.rpicheck.activity.MainActivity - Device cursor rows: 1
22:40:13.814 [main] DEBUG d.e.rpicheck.activity.MainActivity - Spinner item selected: pos=0, id=1
22:40:15.667 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device)
22:40:15.667 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving new instance of all devices.

eidottermihi, 2016-03-30:
Because the app logfile looks normal, I can't really figure out whats going on.
Could you try reinstalling the app?

Offerel, 2016-03-30:
I have done deinstalling and reinstalling the app before i made this post :-) For me it looks like, that the App could successfully pull all the required infos from the Server. But cant parse the Data. I have seen that there was some updates in the last time for some Packages on Debian/Raspbian. Maybe this updates break something in the returned data format what the app cant read.

Is there a way, where i can pull the needed information, module for module? Maybe only one information is in the wrong format.

eidottermihi, 2016-04-21:
You could try to get a full Logcat while the app is crashing, so that we get the stacktrace for the crash.

closed 2016-08-08 by eidottermihi