eidottermihi/rpicheck #127: Support PINE64

PanderMusubi, 2016-09-03:
Please, support PINE64. Contact me for testing.

Phoenix1747, 2017-08-28:
Why would you need support for your Pine while this app is clearly built to work only with a Raspberry Pi?

PanderMusubi, 2017-08-28:
The PINE64 platform is eventually not following a path alongside PRi, as was expected by some. Then reusing code would be beneficial. Part of the functionality offered here is now solved for PINE64 with some bash scripts. So, I will close this is this is no longer interesting to explore further. Thanks for your question.

PanderMusubi, 2017-08-28:
Just out of curiosity, what part of the code here is RPi-specific?

closed 2017-08-28 by PanderMusubi

Phoenix1747, 2017-08-28:
It's RasPi Check not PINE64 Check, why would the dev invest time to support PINE

Phoenix1747, 2017-08-28:
But arguing about this now is really stupid, so yeah

PanderMusubi, 2017-08-28:
Not to argue. If it is only in the name/branding, then I know that.

eidottermihi, 2017-08-28:
The app relies on vcgencmd for example. This is part of the Raspberry Pi
Firmware. So most parts of the Overclocking section wouldn't work on PINE64.

I'm closing this now.