eidottermihi/rpicheck #131: Feature request: Be able to set the interval for status check on widget.

spiral6, 2016-09-04:
I don't want the widget checking every 3 min, it is pretty impactful on my battery. I would like it if we could set the interval to every 5, 10, 30, or 60 min, or once a day, or not at all and only update manually.

Phoenix1747, 2016-09-04:
I don't know what your problem is, but when I place a widget I can choose between following intervals: 5min, 10min, 30min, 1hr, 2hr, 8hr, 1day, own interval.

eidottermihi, 2016-09-04:
What @Phoenix1747 said

closed 2016-09-04 by eidottermihi

spiral6, 2016-09-07:
Ah. Thanks for letting me know. There should be a way to change it after you placed the widget, otherwise you have to remove it and place it again to get those options.

Phoenix1747, 2016-09-07:
It's not that much work to place it again and adjust all settings :P