eidottermihi/rpicheck #135: Dependency Error for io.freefair.android-color:material...

madhucwd, 2017-01-24:
I'm getting the below dependency error for both material and utils. Please suggest, thanks.
"
Warning:Module 'io.freefair.android-colors:material:1.2.0' depends on one or more Android Libraries but is a jar
Warning:Module 'io.freefair.android-util:function:1.1.1' depends on one or more Android Libraries but is a jar
Warning:Module 'io.freefair.android-util:preference:1.1.1' depends on one or more Android Libraries but is a jar
Warning:Module 'io.freefair.android-colors:material:1.2.0' depends on one or more Android Libraries but is a jar
Warning:Module 'io.freefair.android-util:function:1.1.1' depends on one or more Android Libraries but is a jar
Warning:Module 'io.freefair.android-util:preference:1.1.1' depends on one or more Android Libraries but is a jar
Warning:[options] bootstrap class path not set in conjunction with -source 1.7
Error:Execution failed for task ':app:prepareDebugDependencies'.
> Dependency Error. See console for details.
"

larsgrefer, 2017-01-27:
This is an error on @jitpack-io.

Somehow the `pom`s of `io.freefair.android-colors:material:1.2.0` etc. got overwritten.

The original `pom` pointed to the `aar`-file, but the new, wrong `pom` points to a wrong `jar` file.

larsgrefer, 2017-02-28:
see #137 

I think this issue can be closed

closed 2017-03-03 by eidottermihi