eidottermihi/rpicheck #137: Make project failed!

JuwanZhu, 2017-02-22:
Import rpicheck project in Android studio 2.2.3, all is latest! But when Make project:
[image]

```
/Users/apple/Juwan/rpicheck/app/build.gradle
Error:Module 'io.freefair.android-util:function:1.1.1' depends on one or more Android Libraries but is a jar
Error:Module 'io.freefair.android-util:preference:1.1.1' depends on one or more Android Libraries but is a jar
```
How can I solve this problem, any help is welcome!

larsgrefer, 2017-02-22:
Hi @JuwanZhu ,

this is a problem with @jitpack which delivers a `jar` file instead of the correct `aar` file

JuwanZhu, 2017-02-23:
Hi @larsgrefer, Thank you for your prompt reply.

I change following code in build.gradle :
```
compile 'io.freefair.android-injection:injection:1.0.0-rc3'
compile 'io.freefair.android-util:preference:1.1.1'
```
TO

```
compile 'io.freefair.android-injection:injection:1.0.0-rc3@aar'
compile 'io.freefair.android-util:preference:1.1.1@aar'
```

---
So make project success, But when I run rpicheck app, it crashed because of the following error:

> E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: de.eidottermihi.raspicheck, PID: 1550
                                                 java.lang.NoClassDefFoundError: Failed resolution of: Lio/freefair/android/util/logging/AndroidLogger;
                                                     at io.freefair.android.injection.helper.RClassHelper.<clinit>(RClassHelper.java:15)
                                                     at io.freefair.android.injection.injector.ActivityInjector.<init>(ActivityInjector.java:14)
                                                     at io.freefair.android.injection.app.InjectionAppCompatActivity.onCreate(InjectionAppCompatActivity.java:34)
                                                     at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(`MainActivity.java:154`)
                                                     at android.app.Activity.performCreate(Activity.java:6303)
                                                     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1108)
                                                     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2376)
                                                     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2483)
                                                     at android.app.ActivityThread.access$900(ActivityThread.java:153)
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349)
                                                     at android.os.Handler.dispatchMessage(Handler.java:102)
                                                     at android.os.Looper.loop(Looper.java:148)
                                                     at android.app.ActivityThread.main(ActivityThread.java:5438)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:739)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:629)
                                                  `Caused by: java.lang.ClassNotFoundException: Didn't find class "io.freefair.android.util.logging.AndroidLogger" on path: DexPathList[[zip file ---`"/data/app/de.eidottermihi.raspicheck-1/base.apk"],nativeLibraryDirectories=[/data/app/de.eidottermihi.raspicheck-1/lib/arm, /vendor/lib, /system/lib]]
                                                     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
                                                     at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
                                                     at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
                                                     at io.freefair.android.injection.helper.RClassHelper.<clinit>(RClassHelper.java:15) 
                                                     at io.freefair.android.injection.injector.ActivityInjector.<init>(ActivityInjector.java:14) 
                                                     at io.freefair.android.injection.app.InjectionAppCompatActivity.onCreate(InjectionAppCompatActivity.java:34) 
                                                     at de.eidottermihi.rpicheck.activity.MainActivity.onCreate(**MainActivity.java:154**) 
                                                     at android.app.Activity.performCreate(Activity.java:6303) 
                                                     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1108) 
                                                     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2376) 
                                                     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2483) 
                                                     at android.app.ActivityThread.access$900(ActivityThread.java:153) 
                                                     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349) 
                                                     at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                     at android.os.Looper.loop(Looper.java:148) 
                                                     at android.app.ActivityThread.main(ActivityThread.java:5438) 
                                                     at java.lang.reflect.Method.invoke(Native Method) 
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:739) 
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:629) 
[... cut ...]
                                                     at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
                                                     at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
                                                     		... 17 more
                                                  Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available

English is not my mother tongue; please excuse any errors on my part

larsgrefer, 2017-02-23:
Hello @JuwanZhu ,

I've already contacted @jitpack-io. They found the error on their side and will deploy a fix soon.
So you should be able to use the original dependency lines.

About your second error: When you use the `@aar` suffix in the dependency declarations, gradle stops resolving the transitive dependencies. `io.freefair.android-injection:injection:1.0.0-rc3` has itself a dependency on `io.freefair.android-util:logging` which was missing in your `apk` when using the `@aar` notation.

JuwanZhu, 2017-02-28:
@larsgrefer Thank you for you help

closed 2017-02-28 by JuwanZhu