eidottermihi/rpicheck #140: Launch Hyperion

bite-your-idols, 2017-05-10:
Maybe this is a silly question, but if I run the hyperion start command it does not work:
`/storage/hyperion/bin/hyperiond.sh /storage/.config/hyperion.config.json > /storage/logfiles/hyperion.log 2>&1 &`

Also I have a [switch script](https://github.com/bite-your-idols/script.hyperion.switch/blob/master/bin/switch.sh) to start/end this process and if I call this script (` sh /path/to/script.sh`) it executes all actions but the same hyperion launch one.

Using sh in any other PC or in the same smartphone with another ssh apps everything works ok.

any clue?

eidottermihi, 2017-05-25:
Hi, please enable Debug mode in settings and post the log here.

bite-your-idols, 2017-05-25:
Here is the log, I exec 2 commands, first hyperion start command and noting happens, then I exec switch script start command and it shows kodi notification but it didn't start the process.

21:04:52.194 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
21:05:02.445 [main] DEBUG d.e.r.activity.CustomCommandActivity - onCreate: get currentDevice out of intent.
21:05:02.446 [main] DEBUG d.e.r.activity.CustomCommandActivity - Setting activity title for device.
21:05:02.446 [main] DEBUG d.e.r.activity.CustomCommandActivity - Initializing ListView
21:05:02.450 [main] DEBUG d.e.r.activity.CustomCommandActivity - onResume() - retrieving lastCommandId.
21:05:02.451 [main] DEBUG d.e.r.activity.CustomCommandActivity - lastCommandId=13
21:05:02.946 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device)
21:05:02.948 [main] DEBUG d.e.rpicheck.activity.MainActivity - Adding current device to all devices.
21:05:06.543 [main] DEBUG d.e.r.activity.CustomCommandActivity - Command pos 10 clicked. Item _id = 16.
21:05:06.545 [main] DEBUG d.e.r.activity.CustomCommandActivity - Opening command dialog.

21:05:06.591 [AsyncTask #20] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - New RaspiQuery for host: 192.168.0.14
21:05:06.591 [AsyncTask #20] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host: 192.168.0.14 on port 22.
21:05:06.595 [AsyncTask #20] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification.
21:05:08.016 [AsyncTask #20] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Running custom command: /storage/hyperion/bin/hyperiond.sh /storage/.config/hyperion.config.json > /storage/logfiles/hyperion.log 2>&1 &
21:05:08.066 [AsyncTask #20] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Output of '/storage/hyperion/bin/hyperiond.sh /storage/.config/hyperion.config.json > /storage/logfiles/hyperion.log 2>&1 &': 
21:05:08.070 [AsyncTask #20] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Disconnecting from host 192.168.0.14.
21:05:14.033 [main] DEBUG d.e.r.activity.CustomCommandActivity - onPause() - saving lastCommandId=16
21:05:16.710 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device)
21:05:16.712 [main] DEBUG d.e.rpicheck.activity.MainActivity - Adding current device to all devices.
21:05:45.498 [main] DEBUG d.e.r.activity.CustomCommandActivity - onCreate: get currentDevice out of intent.
21:05:45.499 [main] DEBUG d.e.r.activity.CustomCommandActivity - Setting activity title for device.
21:05:45.499 [main] DEBUG d.e.r.activity.CustomCommandActivity - Initializing ListView
21:05:45.503 [main] DEBUG d.e.r.activity.CustomCommandActivity - onResume() - retrieving lastCommandId.
21:05:45.503 [main] DEBUG d.e.r.activity.CustomCommandActivity - lastCommandId=16
21:05:45.976 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device)
21:05:45.979 [main] DEBUG d.e.rpicheck.activity.MainActivity - Adding current device to all devices.
21:05:57.307 [main] DEBUG d.e.r.activity.CustomCommandActivity - Command pos 5 clicked. Item _id = 11.
21:05:57.309 [main] DEBUG d.e.r.activity.CustomCommandActivity - Opening command dialog.
21:05:57.350 [AsyncTask #22] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - New RaspiQuery for host: 192.168.0.14
21:05:57.351 [AsyncTask #22] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host: 192.168.0.14 on port 22.
21:05:57.356 [AsyncTask #22] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification.
21:05:57.713 [AsyncTask #22] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Running custom command: sh /storage/.kodi/addons/script.hyperion.switch/bin/switch.sh
21:05:57.956 [AsyncTask #22] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Output of 'sh /storage/.kodi/addons/script.hyperion.switch/bin/switch.sh': Sending action: Notification(Hyperion,Enabled,2000,/storage/.kodi/addons/script.hyperion.switch/icon.png)

21:05:57.959 [AsyncTask #22] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Disconnecting from host 192.168.0.14.
21:06:17.314 [main] DEBUG d.e.r.activity.CustomCommandActivity - onPause() - saving lastCommandId=11
21:06:17.340 [main] DEBUG d.e.r.activity.CustomCommandActivity - Writing currentDevice in outState.

After that I realized that if I call directly to kodi addon that calls that switch script it works:
>kodi-send --action="RunAddon(script.hyperion.switch)

So I have solve my problem but it is strange why it didn't exec the script itself...

closed 2017-12-18 by eidottermihi