eidottermihi/rpicheck #144: WLAN interface incorrectly listed as unused/disconnected

astro7467, 2017-08-25:
When LAN and WLAN both connected, only LAN info reported. Can connect via WLAN and still being reported as down (or no data shown).

eidottermihi, 2017-08-25:
Hi, please enable debug logging and post the log here. Thanks.

astro7467, 2017-08-25:
Log report with debug:
'07:51:22.827 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device) 07:51:22.835 [main] DEBUG d.e.rpicheck.activity.MainActivity - Adding current device to all devices. 07:51:30.088 [main] DEBUG d.e.rpicheck.activity.MainActivity - Load average preference: FIVE_MINUTES 07:51:30.093 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - New RaspiQuery for host: 192.168.1.253 07:51:30.096 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host: 192.168.1.253 on port 22. 07:51:30.113 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification. 07:51:30.602 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Querying vcgencmd... 07:51:30.653 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Checking vcgencmd location: vcgencmd 07:51:30.685 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Path check output: 07:51:30.688 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Found vcgencmd in path: vcgencmd. 07:51:30.858 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Querying core volts... 07:51:31.315 [AsyncTask #7] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Querying firmware version, vcgencmd path=vcgencmd 07:51:31.403 [AsyncTask #7] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Firmware version: 19606343 (clean) (release) 07:51:31.404 [AsyncTask #7] INFO d.e.r.s.i.queries.LoadAverageQuery - Querying load average for time period FIVE_MINUTES 07:51:31.509 [AsyncTask #7] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: 0.00 0.03 0.02 1/238 6304 07:51:31.510 [AsyncTask #7] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: 5 07:51:31.511 [AsyncTask #7] INFO d.e.r.ssh.impl.queries.UptimeQuery - Querying uptime... 07:51:31.594 [AsyncTask #7] INFO d.e.r.ssh.impl.queries.MemoryQuery - Querying memory information... 07:51:32.012 [AsyncTask #7] DEBUG d.e.r.ssh.impl.queries.MemoryQuery - Using MemAvailable for calculation of free memory. 07:51:32.013 [AsyncTask #7] INFO d.e.r.ssh.impl.queries.SerialNoQuery - Querying serial number... 07:51:32.092 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Querying running processes... 07:51:32.238 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Querying network interfaces... 07:51:32.623 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Found interface eth0. 07:51:32.625 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Found interface wlan0. 07:51:32.626 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Available interfaces: [eth0, wlan0] 07:51:32.627 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Checking carrier of eth0... 07:51:33.029 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - eth0 has a carrier up and running. 07:51:33.031 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Checking carrier of wlan0... 07:51:33.123 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - wlan0 has no carrier. 07:51:33.125 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Querying ip address of interface: 'eth0' 07:51:33.239 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Output of ip query: inet 192.168.1.254/24 brd 192.168.1.255 scope global eth0 07:51:33.241 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - eth0 - IP address: 192.168.1.254. 07:51:33.242 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Querying disk usage... 07:51:33.345 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Disks: [DiskUsageBean [fileSystem=/dev/root, size=3.2G, used=2.7G, available=407M, usedPercent=87%, mountedOn=/], DiskUsageBean [fileSystem=devtmpfs, size=484M, used=0, available=484M, usedPercent=0%, mountedOn=/dev], DiskUsageBean [fileSystem=tmpfs, size=98M, used=1.3M, available=97M, usedPercent=2%, mountedOn=/run], DiskUsageBean [fileSystem=tmpfs, size=5.0M, used=0, available=5.0M, usedPercent=0%, mountedOn=/run/lock], DiskUsageBean [fileSystem=tmpfs, size=196M, used=0, available=196M, usedPercent=0%, mountedOn=/run/shm], DiskUsageBean [fileSystem=/dev/mmcblk0p1, size=56M, used=23M, available=34M, usedPercent=40%, mountedOn=/boot], DiskUsageBean [fileSystem=/dev/sda1, size=114G, used=2.4G, available=112G, usedPercent=3%, mountedOn=/media/8515a151-26b4-44b6-b1b6-85251ee6805a], DiskUsageBean [fileSystem=/dev/root, size=3.2G, used=2.7G, available=407M, usedPercent=87%, mountedOn=/var/folder2ram/var/lib/rrdcached], DiskUsageBean [fileSystem=/dev/root, size=3.2G, used=2.7G, available=407M, usedPercent=87%, mountedOn=/var/folder2ram/var/tmp], DiskUsageBean [fileSystem=/dev/root, size=3.2G, used=2.7G, available=407M, usedPercent=87%, mountedOn=/var/folder2ram/var/lib/monit], DiskUsageBean [fileSystem=folder2ram, size=489M, used=0, available=489M, usedPercent=0%, mountedOn=/var/tmp], DiskUsageBean [fileSystem=folder2ram, size=489M, used=8.0K, available=489M, usedPercent=1%, mountedOn=/var/lib/monit], DiskUsageBean [fileSystem=folder2ram, size=489M, used=18M, available=471M, usedPercent=4%, mountedOn=/var/lib/rrdcached], DiskUsageBean [fileSystem=/dev/root, size=3.2G, used=2.7G, available=407M, usedPercent=87%, mountedOn=/var/folder2ram/var/lib/php5], DiskUsageBean [fileSystem=/dev/root, size=3.2G, used=2.7G, available=407M, usedPercent=87%, mountedOn=/var/folder2ram/var/log], DiskUsageBean [fileSystem=folder2ram, size=489M, used=310M, available=179M, usedPercent=64%, mountedOn=/var/log], DiskUsageBean [fileSystem=folder2ram, size=489M, used=4.0K, available=489M, usedPercent=1%, moun [... cut ...] 7:51:43.473 [AsyncTask #9] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Querying distribution name... 07:51:43.546 [AsyncTask #9] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Disconnecting from host 192.168.1.254. 07:51:43.564 [AsyncTask #9] DEBUG d.e.rpicheck.activity.SSHQueryTask - Query time: 4044 ms. 07:51:50.555 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device) 07:51:50.557 [main] DEBUG d.e.rpicheck.activity.MainActivity - Adding current device to all devices. '

eidottermihi, 2017-08-27:
The log says:
`07:51:33.031 [AsyncTask #7] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Checking carrier of wlan0... 07:51:33.123 [AsyncTask #7] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - wlan0 has no carrier.`

RaspiCheck just runs `cat /sys/class/net/wlan0/carrier`, which in your case presumably returns '0', which means the link is down (see https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-class-net). Therefore RaspiCheck shows the interface as down.

Seems strange, because if you can connect via WLAN, the carrier should be up and return '1'.

closed 2018-01-14 by eidottermihi