eidottermihi/rpicheck #161: Make 'vcgencmd' optional

eidottermihi, 2017-12-25:
Currently, the app needs `vcgencmd` to work properly. vcgencmd is part of every Raspberry Pi's firmware and is used to get "hardware" information (cpu speed, temperature etc.).

Making the availablity of `vcgencmd` optional would open the app for usage with other linux based distributions and boards (BananaPi etc), as all other functionality is based on standard linux utilities.

Nutomic, 2018-05-19:
This would be awesome!

good4y0u, 2018-11-24:
This would be very useful

clach04, 2018-12-01:
Alternative (and would not require developer support initially) is to write a shell script called `vcgencmd` to fake out a Pi on a non-raspberrry pi system.

good4y0u, 2018-12-02:
> Alternative (and would not require developer support initially) is to write a shell script called `vcgencmd` to fake out a Pi on a non-raspberrry pi system.

That is true and not too hard to do...even using real information

zeevy, 2019-05-26:
Any progress on this?

emma-goldman, 2019-12-08:
I'm running Manjaro ARM on a RPi 4 and got help fro mthe Manjaro community to get vcgencmd working on Manjaro ARM using the source code from here [](https://github.com/raspberrypi/userland)

I have an issue on the Manjaro ARM forum ([](https://forum.manjaro.org/t/manjaro-arm-rpi4-multilib-raspberrypi-firmware/114168)) about this, semi-solved, seeing as I could get some queries to work but not all of them so thought I'd cross-post here.

As for ubuntu there is this package `raspberrypi-firmware` that includes `vcgencmd` (source: [](https://archlinuxarm.org/forum/viewtopic.php?f=60&t=12746))

The last log I got with the debug setting on looks like this:
`00:17:01.417 [main] WARN  d.e.r.activity.SettingsActivity - Enabling debug logging. Be warned that the log file can get huge because of this.
00:17:01.423 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
00:17:13.201 [main] DEBUG d.e.rpicheck.activity.MainActivity - Load average preference: ONE_MINUTE
*bunch of keystore and signature lines omitted*
00:17:13.240 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Initialiazed new RaspiQuery for host aurorapi on port 22
00:17:13.240 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host 'aurorapi' on port '22'.
00:17:13.246 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification.
00:17:14.964 [AsyncTask #26] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Querying vcgencmd...
00:17:15.277 [AsyncTask #26] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Checking vcgencmd location: vcgencmd
00:17:15.305 [AsyncTask #26] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Path check output: usage: vcgencmd [-t] command
send a command to the videocore and print the result.

  -t          time how long the command takes to complete
  -h, --help  show this information

use the command 'vcgencmd commands' to get a list of available commands

exit status:
   0    command completed successfully
  -1    problem with vchi
  -2    videocore returned an error

for further documentation please see
https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md

00:17:15.306 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Found vcgencmd in path: vcgencmd.
00:17:15.452 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Querying core volts...
00:17:15.606 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Querying firmware version, vcgencmd path=vcgencmd
00:17:15.680 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Firmware version: aabb1fb5c19d80db268aeccd67b9f4e0f3d48a2a (clean) (release) (start)
00:17:15.681 [AsyncTask #26] INFO  d.e.r.s.i.queries.LoadAverageQuery - Querying load average for time period ONE_MINUTE
00:17:15.763 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: cat: command not found
00:17:15.764 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.764 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.764 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: cat: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: grep: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: grep: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: grep: command not found
00:17:15.766 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: wc: command not found
00:17:15.766 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: wc: command not found
00:17:15.766 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: wc: command not found
00:17:15.766 [AsyncTask #26] ERROR d.e.r.s.i.queries.LoadAverageQuery - Expected a different output of command: cat /proc/loadavg; cat /proc/stat | grep cpu | wc -l
00:17:15.767 [AsyncTask #26] ERROR d.e.r.s.i.queries.LoadAverageQuery - Actual output was: bash: cat: command not found
bash: cat: command not found
bash: grep: command not found
bash: wc: command not found

00:17:15.767 [AsyncTask #26] INFO  d.e.r.ssh.impl.queries.UptimeQuery - Querying uptime...
00:17:15.839 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.UptimeQuery - Skipping line: 
00:17:15.839 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.UptimeQuery - Expected a different output of command: cat /proc/uptime
00:17:15.840 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.UptimeQuery - Actual output was: 
00:17:15.840 [AsyncTask #26] INFO  d.e.r.ssh.impl.queries.MemoryQuery - Querying memory information...
00:17:15.915 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.MemoryQuery - Expected a different output of command: cat /proc/meminfo | tr -s " "
00:17:15.915 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.MemoryQuery - Output was : 
00:17:15.916 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.MemoryQuery - Expected a different output of command: cat /proc/meminfo | tr -s " "
[... cut ...]
00:17:16.453 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - System time: 
00:17:16.454 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Disconnecting from host a.`

There's also some screenies on the Manjaro ARM forum and some more details if anyone else is running that distro and want a somewhat functional raspicheck.
It's strange that I can get some info but not others... I'm thinkin it might have to do with the usergroup video and its permissions?

As for running raspicheck with other hardware I'm not really sure how.

emma-goldman, 2019-12-08:
Hi,
I got this somewhat working with an RPi4 running Manjaro ARM using the help of [the manjaro community](https://forum.manjaro.org/t/manjaro-arm-rpi4-multilib-raspberrypi-firmware/114168/4). Over on that forum there are some screenshots of the info I managed to parse and also I guess what I didn't.
I basically compiled the @raspberrypi /userland source code locally which gave me vcgencmd and then set the appropriate variable. There's apparently precompiled binaries for both ubuntu and archlinux but at least the arch one is only 32bit and Manjaro ARM does not multilib so...

My latest log from raspicheck looks like this:

```
00:17:01.417 [main] WARN  d.e.r.activity.SettingsActivity - Enabling debug logging. Be warned that the log file can get huge because of this.
00:17:01.423 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
00:17:13.201 [main] DEBUG d.e.rpicheck.activity.MainActivity - Load average preference: ONE_MINUTE
*bunch of keystore and signature lines omitted*
00:17:13.240 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Initialiazed new RaspiQuery for host aurorapi on port 22
00:17:13.240 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host 'aurorapi' on port '22'.
00:17:13.246 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification.
00:17:14.964 [AsyncTask #26] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Querying vcgencmd...
00:17:15.277 [AsyncTask #26] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Checking vcgencmd location: vcgencmd
00:17:15.305 [AsyncTask #26] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Path check output: usage: vcgencmd [-t] command
send a command to the videocore and print the result.

  -t          time how long the command takes to complete
  -h, --help  show this information

use the command 'vcgencmd commands' to get a list of available commands

exit status:
   0    command completed successfully
  -1    problem with vchi
  -2    videocore returned an error

for further documentation please see
https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md

00:17:15.306 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Found vcgencmd in path: vcgencmd.
00:17:15.452 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Querying core volts...
00:17:15.606 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Querying firmware version, vcgencmd path=vcgencmd
00:17:15.680 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Firmware version: aabb1fb5c19d80db268aeccd67b9f4e0f3d48a2a (clean) (release) (start)
00:17:15.681 [AsyncTask #26] INFO  d.e.r.s.i.queries.LoadAverageQuery - Querying load average for time period ONE_MINUTE
00:17:15.763 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: cat: command not found
00:17:15.764 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.764 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.764 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: cat: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: cat: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: grep: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: grep: command not found
00:17:15.765 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: grep: command not found
00:17:15.766 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Checking line: bash: wc: command not found
00:17:15.766 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: wc: command not found
00:17:15.766 [AsyncTask #26] DEBUG d.e.r.s.i.queries.LoadAverageQuery - Skipping line: bash: wc: command not found
00:17:15.766 [AsyncTask #26] ERROR d.e.r.s.i.queries.LoadAverageQuery - Expected a different output of command: cat /proc/loadavg; cat /proc/stat | grep cpu | wc -l
00:17:15.767 [AsyncTask #26] ERROR d.e.r.s.i.queries.LoadAverageQuery - Actual output was: bash: cat: command not found
bash: cat: command not found
bash: grep: command not found
bash: wc: command not found

00:17:15.767 [AsyncTask #26] INFO  d.e.r.ssh.impl.queries.UptimeQuery - Querying uptime...
00:17:15.839 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.UptimeQuery - Skipping line: 
00:17:15.839 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.UptimeQuery - Expected a different output of command: cat /proc/uptime
00:17:15.840 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.UptimeQuery - Actual output was: 
00:17:15.840 [AsyncTask #26] INFO  d.e.r.ssh.impl.queries.MemoryQuery - Querying memory information...
00:17:15.915 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.MemoryQuery - Expected a different output of command: cat /proc/meminfo | tr -s " "
00:17:15.915 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.MemoryQuery - Output was : 
00:17:15.916 [AsyncTask #26] ERROR d.e.r.ssh.impl.queries.MemoryQuery - Expected a different output of command: cat /proc/meminfo | tr -s " "
[... cut ...]
00:17:16.374 [AsyncTask #26] ERROR d.e.rpicheck.ssh.impl.RaspiQuery - Output of cat /etc/*-release | grep PRETTY_NAME: 

00:17:16.374 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - Querying system time via 'date --rfc-2822'.
00:17:16.453 [AsyncTask #26] DEBUG d.e.r.ssh.impl.queries.FirmwareQuery - System time: 
00:17:16.454 [AsyncTask #26] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Disconnecting from host a.
```

I know this doesn't help directly with ppl running different boards but at least ppl running other distros might find this useful. 
Cheers.

clach04, 2019-12-08:
@emma-goldman interesting, based on:

    bash: cat: command not found
    bash: grep: command not found
    bash: wc: command not found

 It looks like porting to another platform needs:

* `vcgencmd`
* `bash`
* `cat`
* `grp`
* `wc`

good4y0u, 2019-12-08:
I haven't had time to look into this, life gets in the way of all dreams, but the whole point of this is to get a similar way to get device data as vcgencmd, which can be built and installed on many alternative systems, but not on ALL systems.
 @emma-goldman is running into a similar issue to the one I was having .  

> It looks like porting to another platform needs:
> vcgencmd
> bash
> cat
> grep
> wc

"bash,cat,grep,wc" should be on at least the vast majority of systems, especially bash..otherwise most of what is being worked on here would not work. This is less of the problem. 

What would be something would be to make it optional and have another way to get the information for the app, it is heavily reliant currently on using the vcgencmd to query information about the device. Which is 100% understandable given its for RPI's which by default use vcgencmd . 

I'm glad there are still people interested in this.

clach04, 2020-01-24:
So I ended up have a need (well, maybe want is more accurate :-p) for this for a rock64 SBC. It runs linux so everything works great, other than the top part of the stats. This is what I came up with.

Create `/usr/bin/vcgencmd` and `chmod a+x` it.

    #!/bin/sh
    # try and keep this pure bourne shell
    # minimal clone of raspberry tool vcgencmd - for use with Android rpicheck and rock64 SBC

    command=$1
    case ${1} in
            measure_clock)
                    case ${2} in
                            arm)
                                    # awk is probably overkill....
                                    /usr/local/sbin/rock64_health.sh | awk '/CPU freq/{print "frequency(45)=" $4 * 1000000}'
                                    #value=`/usr/local/sbin/rock64_health.sh | awk '/CPU freq/{print $4 * 1000000}'`
                                    #echo 'frequency(45)='${value}
                                    exit
                            ;;
                            core)
                                    value=0  # TODO / FIXME
                                    echo 'frequency(1)='${value}
                                    exit
                            ;;
                    # TODO anything else thrown an error/debug
                    esac
                    exit
                    ;;
            measure_temp)
                    # awk is probably overkill....
                    #/usr/local/sbin/rock64_health.sh | awk '/SoC Temp/{print $4 "\\\'C"}'
                    value=`/usr/local/sbin/rock64_health.sh | awk '/SoC Temp/{print $4}'`
                    echo 'temp='${value}"'C"
                    exit
                    ;;
            measure_volts)
                    case ${2} in
                            core)
                                    value=0  # TODO / FIXME
                                    echo 'volt='${value}'.0000V'
                                    exit
                            ;;
                    # TODO anything else thrown an error/debug
                    esac
                    ;;
            version)
                    echo 'Nov  4 2018 16:31:07'
                    echo 'Copyright (c) 2012 rock64'
                    echo 'version rock64_TODO (clean) (release)'
                    exit
                    ;;
    esac

    #!/bin/sh
    # try and keep this pure bourne shell
    # vcgencmd - clone

    # on this device /var/log is ramfs folder2ram
    LOG=/var/log/vcgencmd.log

    date >> ${LOG}
    echo ${*} >> ${LOG}
    echo '=============' >> ${LOG}

The last few lines are to debug when unhandled parameters are requested.

clach04, 2020-01-25:
Posted a version to https://github.com/clach04/rock64_vcgencmd

Thatoo, 2020-02-27:
Would this script work on a x86 server?

clach04, 2020-08-29:
@Thatoo I just renamed the project to https://github.com/clach04/fake_vcgencmd

It works for me under Ubuntu x86, as well as Rock64 legacy, and Armbian. There is a chance small tweaks are needed but for most platforms that expose the typical:

 * `/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq`
* `/sys/class/thermal/thermal_zone0/temp`

devices for lookup it should work, and its easy tune to specific hardware.

clach04, 2022-03-19:
@eidottermihi I don't have an Android SDK /enthusiasm to work on a code change for this support :-p. I can think of three possible directions for this ticket:

1. remove hard dependency - most amount of work
2. add a check to RasPiCheck for presence of vcgencmd and if missing "do something" (which could be a host of things, including generating an almost empty shell script that would return 0 for all the things RasPiCheck current checks for)
3. Document what RasPiCheck expects so people can implement their own vcgencmd replacement (for example use https://github.com/clach04/fake_vcgencmd) - no code changes, hopefully minor doc work (perhaps add an FAQ?)

I do have energy for number 3 above if that's something you are open to? I could knock something up in https://github.com/eidottermihi/rpicheck/tree/gh-pages and open a PR (I could attempt #228 at the same time)

mepanko91, 2023-03-25:
Modified clach04 script
https://github.com/clach04/fake_vcgencmd 
little bit to get highest core temp instead of motherboard temp

```
        measure_temp)
                # awk is probably overkill....
                value=`cat /sys/devices/platform/coretemp.0/hwmon/hwmon*/temp*_input | sort -n | tail -1 | awk '{printf "%.3f\n", $1 / 1000}'`
```
Many thanks for script. Cheers

PanderMusubi, 2023-04-21:
I think this can be closed as using fake_vcgencmd solves it.

PanderMusubi, 2023-04-22:
This app can now also be used on non-Raspberry Pi boards with Armbian operating system.

PanderMusubi, 2023-04-24:
I will make a PR for it.

PanderMusubi, 2024-05-07:
Link to vcgencmd has been added to the README and website, so I think this issue can be closed.

clach04, 2024-05-11:
Looks like this was essentially taken care of in commit https://github.com/eidottermihi/rpicheck/commit/188808be5dacdd94aa44c397b7c1ff6aaa2e569f (basically option 3 from comment https://github.com/eidottermihi/rpicheck/issues/161#issuecomment-1073128501) it's not completely documented what is needed but the https://github.com/clach04/fake_vcgencmd script basically does.

clach04, 2024-05-11:
Thanks to @PanderMusubi for taking care of this (I was travelling and didn't see this until I got back and PanderMusubi / @eidottermihi  took care of it by the time I did see it :-) )

closed 2024-05-14 by eidottermihi