eidottermihi/rpicheck #178: Unbuildable by F-Droid

gsantner, 2018-04-08:
Hi there, I saw that recently a jar was checked in (for ssh module). This makes the app unbuildable by F-Droid as they only allow to build from source or from mvm repos that provide the source for the jar together with the jar.

Is there any way the jar could be removed and instead the lib be loaded from e.g. mavencentral or jitpack?

Btw, F-Droid is currently on the state of `1.8.0`, don't know what happened since that. <https://gitlab.com/fdroid/fdroiddata/blob/master/metadata/de.eidottermihi.raspicheck.txt>

Thank you!

eidottermihi, 2018-04-10:
Thanks for reminding, the jar was a temporarily workaround for as long as sshj version >= 0.24 wasn't released. As this is no longer the case, I will switch to the official release (and will look into F-Droid metadata).

eidottermihi, 2018-09-15:
fixed in 0e76097a090a491dc358f9c3412f33f862ead0da, v1.8.10

closed 2018-09-15 by eidottermihi

PanderMusubi, 2020-03-25:
@eidottermihi see https://github.com/eidottermihi/rpicheck/issues/195 for metadata

Other question, see can you add versionName as is mentioned in https://gitlab.com/fdroid/fdroiddata/-/issues/1988#note_311073711 so that F-Droid can build new releases automatically?