eidottermihi/rpicheck #181: vcgencmd installed and ssh user in 'video' group but not working.

chsims1, 2018-05-20:
Hi,

as the title says (/usr/bin/vcgencmd).  I can run 'sudo vcgencmd measure_temp' successfully from a ssh session, but app complains that vcgencmd can't be found or ssh user is not member of video group (both incorrect).

There is no log file as this distro (Volumio) runs with squashfs in RAM to minimise sd card writes. Presumably the absence of the log file wouldn't cause a failure??

eidottermihi, 2018-05-20:
Hi,
the app runs vcgencmd without sudo. So you have to make sure your user has
the neccesary rights to execute vcgencmd without sudo.

chsims1, 2018-05-20:
Ah ok, it failed without being sudo. I'll have a look when I can access the device. Thanks

ChristopherRush, 2018-08-06:
Had a similar Issue with Volumio, when adding user to group video `sudo adduser volumio video` it did not resolve the issue. The culprit was `/dev/vchiq`.

`ls -l /dev/vchiq` returns:
`crw------- 1 root root 246, 0 Jun 17 22:13 /dev/vchiq`
Not:
`crw-rw---- 1 root video 246, 0 Jun 16 23:28 /dev/vchiq`

You need to run the following:
`sudo chmod g+rw /dev/vchiq`
`sudo chgrp video /dev/vchiq`

closed 2019-12-08 by eidottermihi