eidottermihi/rpicheck #183: Cannot connect with priv key w. pass (received disconnect from phone).

kl0z, 2018-06-24:
Hi I'm trying to connect as the title says but get "SSH authetication failed" on the phone. On auth.log I get:

```
Jun 24 20:32:14 nodomusikal sshd[2404]: Connection from (phone IP) port 54543 on 192.168.1.128 port 50138
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: Client protocol version 2.0; client software version SSHJ_0.23.0_ANDROID_COMPAT_2_SNAPSHOT
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: no match: SSHJ_0.23.0_ANDROID_COMPAT_2_SNAPSHOT
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: Enabling compatibility mode for protocol 2.0
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: Local version string SSH-2.0-OpenSSH_6.7p1 Raspbian-5+deb8u4
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: permanently_set_uid: 107/65534 [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: list_hostkey_types: ssh-rsa,ssh-dss,ecdsa-sha2-nistp256,ssh-ed25519 [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: SSH2_MSG_KEXINIT sent [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: SSH2_MSG_KEXINIT received [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: kex: client->server aes128-ctr hmac-sha1 none [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: kex: server->client aes128-ctr hmac-sha1 none [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: expecting SSH2_MSG_KEX_ECDH_INIT [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: SSH2_MSG_NEWKEYS sent [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: expecting SSH2_MSG_NEWKEYS [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: SSH2_MSG_NEWKEYS received [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: KEX done [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: Received disconnect from (phone IP):  [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: do_cleanup [preauth]
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: monitor_read_log: child log fd closed
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: do_cleanup
Jun 24 20:32:14 nodomusikal sshd[2404]: debug1: Killing privsep child 2405
```

Mi sshd_config is,

```
# Package generated configuration file
# See the sshd_config(5) manpage for details

# What ports, IPs and protocols we listen for
Port 50138

# Use these options to restrict which interfaces/protocols sshd will bind to
#ListenAddress ::
#ListenAddress 0.0.0.0

Protocol 2

# HostKeys for protocol version 2
HostKey /etc/ssh/ssh_host_rsa_key
HostKey /etc/ssh/ssh_host_dsa_key
HostKey /etc/ssh/ssh_host_ecdsa_key
HostKey /etc/ssh/ssh_host_ed25519_key

#Privilege Separation is turned on for security
UsePrivilegeSeparation yes

# Lifetime and size of ephemeral version 1 server key
KeyRegenerationInterval 3600
ServerKeyBits 1024

# Logging
SyslogFacility AUTH
LogLevel DEBUG

# Authentication:
LoginGraceTime 120
PermitRootLogin no
StrictModes yes
MaxAuthTries 15
AllowUsers (...)
#AllowGroups sftpusers

RSAAuthentication yes
PubkeyAuthentication yes
#AuthorizedKeysFile     %h/.ssh/authorized_keys

# Don't read the user's ~/.rhosts and ~/.shosts files
IgnoreRhosts yes
# For this to work you will also need host keys in /etc/ssh_known_hosts
RhostsRSAAuthentication no
# similar for protocol version 2
HostbasedAuthentication no
# Uncomment if you don't trust ~/.ssh/known_hosts for RhostsRSAAuthentication
#IgnoreUserKnownHosts yes

# To enable empty passwords, change to yes (NOT RECOMMENDED)
PermitEmptyPasswords no

# Change to yes to enable challenge-response passwords (beware issues with
# some PAM modules and threads)
ChallengeResponseAuthentication no

# Change to no to disable tunnelled clear text passwords
PasswordAuthentication no

# Kerberos options
#KerberosAuthentication no
#KerberosGetAFSToken no
#KerberosOrLocalPasswd yes
#KerberosTicketCleanup yes

# GSSAPI options
GSSAPIAuthentication no
#GSSAPICleanupCredentials yes
UseDNS no

X11Forwarding no
X11DisplayOffset 10
PrintMotd no
PrintLastLog yes
TCPKeepAlive yes
#UseLogin no

#MaxStartups 10:30:60
#Banner /etc/issue.net

# Allow client to pass locale environment variables
AcceptEnv LANG LC_*
```

I can connect from the phone with other SSH clients without problem. Is there a way I can view the logs generated by your app so I can post them here?

Thank you!

kl0z, 2018-06-26:
Wow, I don't know how I didn't see that... although I'm doing as you are saying but still no logs display. Only message I get is "SSH authentication failed. Please check your credentials." And if I tap on the "view log" option (with debug logging enabled) I'm thrown back to the status screen (the one that displays: overcloking, system, network, ...).

Now I'm quite positive that the credentials are correct because I copy/paste them and tried a lot of times... also, the key is in format PKCS8/OpenSSH and seems to be recognized by the app.

I'm using Huawei P8 lite ALE-L21 with EMUI version 4.0.3 and Android 6.0. Please tell me what I can do to gain further insight about the issue. Is there maybe a log file that I can open with a text editor?

Thank you again.

eidottermihi, 2018-07-06:
@kl0z the log file should be located on your storage, directory `Android/data/de.eidottermihi.rpicheck`

good4y0u, 2018-11-24:
does yours work with a private key without a password?

eidottermihi, 2021-04-30:
Closing due to inactivity.

closed 2021-04-30 by eidottermihi