eidottermihi/rpicheck #185: Can't get past sudo prompts

filthycoding, 2018-09-23:
Hello,

I would like to add the command `sudo apt update` to my commands on the app.

So far, I've tried
```
sudo apt update
%{pi.sudopass}
```
and
```
echo -e "%{pi.sudopass}\n" | sudo apt-get update
```

However, the dialog showing output just gets stuck at "Connection established" and eventually timeouts:
```
Connecting to host _<censored>_  ...
Authenticating with private key and passphrase ...
Connection established.
ERROR - A transport exception occured. Check your connection.
Reason: Timeout expired
ERROR - A transport exception occured. Check your connection.
Reason: Timeout expired
```
The debug log just shows this:
```
16:39:13.018 [AsyncTask #5] INFO d.e.rpicheck.ssh.impl.RaspiQuery - Running custom command: echo -e "<censored password>\n"| sudo apt-get update 16:39:23.362 [AsyncTask #5] ERROR d.e.r.fragment.RunCommandDialog - Exception occured during command execution. Caused by: de.eidottermihi.rpicheck.ssh.impl.RaspiQueryException: A transport exception occured. Check your connection. ... Caused by: java.util.concurrent.TimeoutException: Timeout expired
```

Am I doing something wrong?

eidottermihi, 2018-09-23:
Did you specify a higher timeout value? The default is 60 seconds. I your action (like `apt-get update`) takes longer than 60 seconds, you should increase the timeout for this command.

You can edit the timeout value if you long-press the command in the list and select `Edit`.

eidottermihi, 2018-09-23:
Alternatively, you could also use `nohup` (like suggested in #171 )

filthycoding, 2018-09-23:
I increased the timeout to 36000 seconds. The issue is that, simple enough, nothing happens after an established connection.

eidottermihi, 2018-09-23:
But you still get the Timeout Exception immediately?

filthycoding, 2018-09-23:
If I set the timeout to 500 seconds, a timeout happens after 500 seconds. If I SSH into my Pi normally using my laptop, that command finishes in about 20 seconds. So it doesn't do anything.

eidottermihi, 2018-09-29:
@filthycoding Try to remove the `\n` from the echo, it is not needed and leads to another prompt (possibly this is why the command times out).

filthycoding, 2018-09-29:
Tried it, yields the same results. I am starting to wonder if that has anything to do with me having enabled password feedback on sudo (every character shows an asterisk).