eidottermihi/rpicheck #186: Private key authentication not working

alexmat2on, 2018-10-21:
Hello, I generated new SSH keys from my laptop using `ssh-keygen`. I added it to my Pi using `ssh-copy-id` and tested it correctly from my computer.

I transferred the private key onto my phone, selected `private key` as my authentication method, and selected the private key file which I placed in `/storage/emulated/0/Documents/op3t`

Here is an except from the log:

```
13:36:35.625 [AsyncTask #32] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using private/public key authentication, keyfile: '/storage/emulated/0/Documents/op3t'
13:36:35.939 [AsyncTask #32] ERROR net.schmizz.concurrent.Promise - <<authenticated>> woke to: net.schmizz.sshj.userauth.UserAuthException: Problem getting private key from PKCS8KeyFile{resource=[PrivateKeyFileResource] /storage/emulated/0/Documents/op3t}
13:36:35.945 [AsyncTask #32] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Authentication failed.
net.schmizz.sshj.userauth.UserAuthException: Exhausted available authentication methods
	at net.schmizz.sshj.SSHClient.auth(SSHClient.java:231) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.authPublickey(SSHClient.java:346) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.authPublickey(SSHClient.java:365) ~[na:0.0]
	at de.eidottermihi.rpicheck.ssh.impl.RaspiQuery.connectWithPubKeyAuth(RaspiQuery.java:789) ~[na:0.0]
	at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:93) ~[na:0.0]
	at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:52) ~[na:0.0]
	at android.os.AsyncTask$2.call(AsyncTask.java:333) ~[na:0.0]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) ~[na:0.0]
	at java.lang.Thread.run(Thread.java:764) ~[na:0.0]
Caused by: net.schmizz.sshj.userauth.UserAuthException: Problem getting private key from PKCS8KeyFile{resource=[PrivateKeyFileResource] /storage/emulated/0/Documents/op3t}
	at net.schmizz.sshj.userauth.method.KeyedAuthMethod.putSig(KeyedAuthMethod.java:61) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AuthPublickey.sendSignedReq(AuthPublickey.java:74) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AuthPublickey.handle(AuthPublickey.java:45) ~[na:0.0]
	at net.schmizz.sshj.userauth.UserAuthImpl.handle(UserAuthImpl.java:142) ~[na:0.0]
	at net.schmizz.sshj.transport.TransportImpl.handle(TransportImpl.java:500) ~[na:0.0]
	at net.schmizz.sshj.transport.Decoder.decode(Decoder.java:102) ~[na:0.0]
	at net.schmizz.sshj.transport.Decoder.received(Decoder.java:170) ~[na:0.0]
	at net.schmizz.sshj.transport.Reader.run(Reader.java:59) ~[na:0.0]
Caused by: java.io.IOException: Could not read key pair from: [PrivateKeyFileResource] /storage/emulated/0/Documents/op3t
	at net.schmizz.sshj.userauth.keyprovider.PKCS8KeyFile.readKeyPair(PKCS8KeyFile.java:96) ~[na:0.0]
	at net.schmizz.sshj.userauth.keyprovider.BaseFileKeyProvider.getPrivate(BaseFileKeyProvider.java:75) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.KeyedAuthMethod.putSig(KeyedAuthMethod.java:59) ~[na:0.0]
	... 7 common frames omitted
```

The private key is plain text in the regular format, something like 

```
-----BEGIN PRIVATE KEY-----
blahblahblahblahblahblah...
-----END PRIVATE KEY-----
```

good4y0u, 2018-11-24:
Same issue. 
I can use a terminal app on my phone -> PI, this works using the key's. when I try to use the same private key with this app -> pi it doesnt work. The key is added correctly to the PI.

eidottermihi, 2018-11-28:
Try to store the public key next to your private key on your phone. The public key file must have the same name as the private key file with ".pub" as suffix.

for your setup:
- /storage/emulated/0/Documents/op3t (private key)
- /storage/emulated/0/Documents/op3t.pub (publik key)

good4y0u, 2018-11-29:
19:39:30.411 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Initialiazed new RaspiQuery for host <hostname_redacted> on port <port_redacted>
19:39:30.412 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host <hostname_redacted> on port <port_redacted>
19:39:30.418 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification.
19:39:30.989 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using private/public key authentication, keyfile: '/storage/emulated/0/.ssh/opt3'
19:39:31.122 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Authentication failed.
net.schmizz.sshj.userauth.UserAuthException: Exhausted available authentication methods
	at net.schmizz.sshj.SSHClient.auth(SSHClient.java:231) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.authPublickey(SSHClient.java:346) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.authPublickey(SSHClient.java:365) ~[na:0.0]
	at de.eidottermihi.rpicheck.ssh.impl.RaspiQuery.connectWithPubKeyAuth(RaspiQuery.java:789) ~[na:0.0]
	at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:93) ~[na:0.0]
	at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:52) ~[na:0.0]
	at android.os.AsyncTask$2.call(AsyncTask.java:333) ~[na:0.0]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) ~[na:0.0]
	at java.lang.Thread.run(Thread.java:764) ~[na:0.0]
Caused by: net.schmizz.sshj.userauth.UserAuthException: Problem getting public key from com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile@a5093e
	at net.schmizz.sshj.userauth.method.KeyedAuthMethod.putPubKey(KeyedAuthMethod.java:46) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AuthPublickey.buildReq(AuthPublickey.java:62) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AuthPublickey.buildReq(AuthPublickey.java:81) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AbstractAuthMethod.request(AbstractAuthMethod.java:68) ~[na:0.0]
	at net.schmizz.sshj.userauth.UserAuthImpl.authenticate(UserAuthImpl.java:72) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.auth(SSHClient.java:225) ~[na:0.0]
	... 10 common frames omitted
Caused by: net.schmizz.sshj.common.Buffer$BufferException: Bad item length: 1263035685
	at net.schmizz.sshj.common.Buffer.readString(Buffer.java:397) ~[na:0.0]
	at net.schmizz.sshj.common.Buffer.readString(Buffer.java:412) ~[na:0.0]
	at com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile.readUnencrypted(OpenSSHKeyV1KeyFile.java:190) ~[na:0.0]
	at com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile.readDecodedKeyPair(OpenSSHKeyV1KeyFile.java:107) ~[na:0.0]
	at com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile.readKeyPair(OpenSSHKeyV1KeyFile.java:79) ~[na:0.0]
	at net.schmizz.sshj.userauth.keyprovider.BaseFileKeyProvider.getPublic(BaseFileKeyProvider.java:81) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.KeyedAuthMethod.putPubKey(KeyedAuthMethod.java:44) ~[na:0.0]
	... 15 common frames omitted
19:39:31.128 [AsyncTask #1] ERROR d.e.rpicheck.activity.SSHQueryTask - Could not authenticate on host '<hostname>' and login '<username>'. Check your login/password.
de.eidottermihi.rpicheck.ssh.impl.RaspiQueryException: Could not authenticate on host '<hostname>' and login '<username>'. Check your login/password.
	at de.eidottermihi.rpicheck.ssh.impl.RaspiQuery.connectWithPubKeyAuth(RaspiQuery.java:792) ~[na:0.0]
	at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:93) ~[na:0.0]
	at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:52) ~[na:0.0]
	at android.os.AsyncTask$2.call(AsyncTask.java:333) ~[na:0.0]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) ~[na:0.0]
	at java.lang.Thread.run(Thread.java:764) ~[na:0.0]
Caused by: net.schmizz.sshj.userauth.UserAuthException: Exhausted available authentication methods
	at net.schmizz.sshj.SSHClient.auth(SSHClient.java:231) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.authPublickey(SSHClient.java:346) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.authPublickey(SSHClient.java:365) ~[na:0.0]
	at de.eidottermihi.rpicheck.ssh.impl.RaspiQuery.connectWithPubKeyAuth(RaspiQuery.java:789) ~[na:0.0]
	... 7 common frames omitted
Caused by: net.schmizz.sshj.userauth.UserAuthException: Problem getting public key from com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile@a5093e
	at net.schmizz.sshj.userauth.method.KeyedAuthMethod.putPubKey(KeyedAuthMethod.java:46) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AuthPublickey.buildReq(AuthPublickey.java:62) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AuthPublickey.buildReq(AuthPublickey.java:81) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.AbstractAuthMethod.request(AbstractAuthMethod.java:68) ~[na:0.0]
	at net.schmizz.sshj.userauth.UserAuthImpl.authenticate(UserAuthImpl.java:72) ~[na:0.0]
	at net.schmizz.sshj.SSHClient.auth(SSHClient.java:225) ~[na:0.0]
	... 10 common frames omitted
Caused by: net.schmizz.sshj.common.Buffer$BufferException: Bad item length: 1263035685
	at net.schmizz.sshj.common.Buffer.readString(Buffer.java:397) ~[na:0.0]
[... cut ...]
	at com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile.readUnencrypted(OpenSSHKeyV1KeyFile.java:190) ~[na:0.0]
	at com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile.readDecodedKeyPair(OpenSSHKeyV1KeyFile.java:107) ~[na:0.0]
	at com.hierynomus.sshj.userauth.keyprovider.OpenSSHKeyV1KeyFile.readKeyPair(OpenSSHKeyV1KeyFile.java:79) ~[na:0.0]
	at net.schmizz.sshj.userauth.keyprovider.BaseFileKeyProvider.getPublic(BaseFileKeyProvider.java:81) ~[na:0.0]
	at net.schmizz.sshj.userauth.method.KeyedAuthMethod.putPubKey(KeyedAuthMethod.java:44) ~[na:0.0]
	... 15 common frames omitted

alexmat2on, 2018-12-06:
`ssh-keygen` on Manjaro Linux. I just renamed the file `op3t` instead of the default `id_rsa` because my phone is a OnePlus 3T

smjohnson85, 2018-12-20:
I'm having the same issue. I generated the keys on my Mac with `ssh-keygen` and copied both `id_rsa` and `id_rsa.pub` to the same folder on my phone (S7 edge). When I select the private key and try to connect to my Pi, I get the SSH authentication failed error.

I can connect just fine from my Mac, and from another Ubuntu laptop where I added those same keys using `ssh-add`.

I used a SHA256 4096bit key.

PwOSS, 2019-01-25:
Same here. RSA 4096. I checked the logs but there isn't any useful information. Just auth is wrong which isn't wrong.

kareliot, 2019-04-02:
This problem is affecting me as well which means that the app is basically useless at the moment. Is there any progress? I am happy to help and debug – just let me know what I can do

feisung, 2019-05-26:
Same issue here. I use juice ssh app with a generated key and have no issues. Even tried generating an openssl key and that also results in inability to connect.
First time installing the app and will uninstall because it clearly is useless if I can't auth

assgex, 2019-06-06:
Same issue here. Using JuieSSH

vasyugan, 2019-10-12:
For me it works with an older SSH key but fails with a newer one, both of which are in ~/.ssh/authorized_keys, and of course, both of which work with other ssh clients. I haven't quite figured out what the difference is, whether there are any differences in terms of encryption algorithms used or so.

online-stuff, 2020-03-09:
I wonder if OpenKeychain's API can be used for authentication instead..  Might be easier than fixing this software.  Plus would also give the option of using hardware keys.

AllirionX, 2020-04-03:
Same problem here. Ssh keys were generated from the android device with Termux and copied with ssh-copy-id.

shameempk, 2020-05-10:
I suspect this could be some file permission issue. I don't know how https://github.com/hierynomus/sshj handles ssh key files though.

vasyugan, 2020-05-10:
> I suspect this could be some file permission issue. I don't know how https://github.com/hierynomus/sshj handles ssh key files though.

No, it isn't. For me it works with one key but doesn't work with another (a newer one), which have both identical permissions. The difference seems to be that they use different encryption algorithms, and the app is unable to handle one of them.

ksz16, 2021-04-13:
I think I have found the solution to the problem.
#204

kareliot, 2021-04-14:
Thanks @ksz16 
Unfortunately though, this did not work for me. Following your suggestion, I tried to convert my existing private key to a PEM private key (see https://unix.stackexchange.com/questions/638043/how-to-convert-rfc4716-private-keys-to-pem-private-keys). 
I still get the same error in raspicheck however. (SSH authentication failed).

Has someone else tried converting an existing key?

kareliot, 2021-04-14:
Sorry, my fault. I accidently used the wrong key file.
I can now indeed confirm that the solution/workaround suggested by @ksz16 in #204 works perfectly.
All you have to do is: 
`ssh-keygen -p -f /foldername/keyfilename -m PEM`

ksz16, 2021-04-14:
@kareliot, glad to hear it's working for you too. I came across this solution by accident. I needed to convert a key from OpenSSH to Dropbear (to do this you first need to change the key format to PEM). And then I remembered about the private key issues on RasPi Check. I decided to use the PEM key. And surprisingly it worked.
I think sharing knowledge is very important. I often use hints and solutions that others share online. So I'm glad it helped you and I hope it helps others as well.

eidottermihi, 2021-04-30:
closing this, solution described in #204 (pinned).

closed 2021-04-30 by eidottermihi