eidottermihi/rpicheck #195: Feature image, screenshots and metadata for app stores

PanderMusubi, 2019-09-18:
Please add in a metadata directory a feature image, screenshots (smartphone and tablet) and other metadata (description and changelogs) which will be used automatically by app stores such as F-Droid and Google Play.

Examples can be found at https://github.com/y20k/transistor/tree/master/metadata/en-US for normal metadata and https://github.com/zamojski/TowerCollector/tree/master/fastlane/metadata/android/en-US/images for fastlane.

PanderMusubi, 2023-04-21:
If you continue maintaining this app, I can create a PR for this.

eidottermihi, 2023-05-08:
@PanderMusubi thanks for the PRs! are we good to go for a new release or is something in `metadata` still missing?

PanderMusubi, 2023-05-08:
I have to add metadata here. Will that later this week.