eidottermihi/rpicheck #201: For more security spongycastle -> bouncycastle

Neustradamus, 2020-04-14:
For more security, can you change old spongycastle (based on old bouncycastle) to bouncycastle?
- https://www.bouncycastle.org/
- https://www.bouncycastle.org/releasenotes.html
- http://www.bouncycastle.org/latest_releases.html
- https://www.cvedetails.com/vulnerability-list/vendor_id-7637/Bouncycastle.html

eidottermihi, 2021-04-30:
I'm not sure what the current state of Bouncycastle on Android is. It was already tricky to get it working with spongycastle, so not quite sure where to start first here.

But I'll happily accept PRs for this.

eidottermihi, 2021-05-03:
Seems like SpongyCastle could be replaced with official BouncyCastle according to

https://github.com/rtyley/spongycastle/issues/34 https://github.com/rtyley/spongycastle/issues/34#issuecomment-292687789

commit 34be2278 (2021-05-04): Switch to bouncycastle and drop SpongyCastle (see https://github.com/rtyley/spongycastle/issues/34), fix #201
closed 2021-05-04 by eidottermihi