eidottermihi/rpicheck #204: Private key not working

vlad34, 2020-07-22:
Hi,

I imported my private key in the "modify the raspberry Pi" but when updating the information I get the message "Oops! SSH authentication".

In the "/var/log/auth.log" of the raspberry pi I have as return:

`Jul 22 17:48:24 pi sshd [14190]: Received disconnect from 192.168.1.230 port 45235: 11: [preauth]
Jul 22 17:48:24 pi sshd [14190]: Disconnected from 192.168.1.230 port 45235 [preauth]`

The key is not used ...,

can you help me.

Thanks.

hille721, 2020-11-29:
I have the same problem.

Any Updates about that?

ksz16, 2021-04-13:
Today I found a solution to this problem completely by accident. Until now I tried to use a key generated in this way:
`ssh-keygen -t rsa -b 4096 -f id_rsa -C "you@your_machine"`
Today I tried it like this:
`ssh-keygen -t rsa -b 4096 -m PEM -f id_rsa -C "you@your_machine"`
You can also convert an existing RFC4716 key to a PEM private key.
I hope this helps someone. RasPi Check is a very cool app, but without private key support it was almost useless to me. Now everything works.

shameempk, 2021-04-26:
@ksz16 That worked. Thanks.

eidottermihi, 2021-04-30:
Thanks guys for sorting this out, I'll pin the issue with the resolution:

Use keys in PEM format, for example:

```
ssh-keygen -t rsa -b 4096 -m PEM -f id_rsa -C "you@your_machine"
```

vlad34, 2021-05-02:
Works, Thanks! :) 

I disabled the connection ssh by password ;) , finally!

closed 2021-05-02 by vlad34