eidottermihi/rpicheck #211: Feature request: PiCore Support

MicPra, 2021-04-24:
Great app to control all my Pi's. Just can't get my Pi with TinyCore/PiCore running. I've installed the package for vcgencmd and it's working so far - but it needs sudo -s elevation. Could you make it possible that RasPi Check issue a "sudo -s" as the first command?

eidottermihi, 2021-04-30:
Please try to add your user to group `video`, then it should work without sudo elevation.

for example:
```
sudo usermod -G video pi
```

MicPra, 2021-04-30:
Hi Michael,
for sure - that's what I did so far before opening this issue. Because it's piCore:
`sudo usermod -aG video tc`
result:
`tc@piCorePlayer:~$ id`
`uid=1001(tc) gid=50(staff) groups=44(video),50(staff)`
But still.. 
`tc@piCorePlayer:~$ vcgencmd`
`VCHI initialization failed`

`sudo -s vcgencmd measure_temp` works as a custom command with RasPi Check fine!

Thanks a lot for your reply and investigation

eidottermihi, 2021-05-02:
@MicPra could you try to "alias" around the issue, aka give your SSH user an alias like

```
alias vcgencmd=sudo -s vcgencmd
```

MicPra, 2021-05-02:
@eidottermihi , this is a hint into the right direction. I can use 'vcgencmd w/o elevated rights from the shell.

tc@piCorePlayer:~$ alias vcgencmd="sudo -s vcgencmd"

tc@piCorePlayer:~$ vcgencmd commands

commands="vcos, ap_output_control, ap_output_post_processing, vchi_test_init, vchi_test_exit, vctest_memmap, vctest_start, vctest_stop, vctest_set, vctest_get, pm_set_policy, pm_get_status, pm_show_stats, pm_start_logging, pm_stop_logging, version, commands, set_vll_dir, set_backlight, set_logging, get_lcd_info, arbiter, cache_flush, otp_dump, test_result, codec_enabled, get_camera, get_mem, measure_clock, measure_volts, enable_clock, scaling_kernel, scaling_sharpness, get_hvs_asserts, get_throttled, measure_temp, get_config, hdmi_ntsc_freqs, hdmi_adjust_clock, hdmi_status_show, hvs_update_fields, pwm_speedup, force_audio, hdmi_stream_channels, hdmi_channel_map, display_power, read_ring_osc, memtest, dispmanx_list, get_rsts, schmoo, render_bar, disk_notify, inuse_notify, sus_suspend, sus_status, sus_is_enabled, sus_stop_test_thread, egl_platform_switch, mem_validate, mem_oom, mem_reloc_stats, hdmi_cvt, hdmi_timings, readmr, pmicrd, pmicwr, bootloader_version, bootloader_config, file"

tc@piCorePlayer:~$

Unfortunately, I didn't get it to make the alias permanent with PiCore. The command 'backup' did not make it permanent. I'm not sure if this is the reason, but I still get this (!) Oops! The tool 'vcgencmd' was not found...

Do you've a further/deeper check in RasPi Config?

eidottermihi, 2021-05-02:
Try setting the alias permanently, like in ~/.profile or ~/.bash_profile, so the alias will be applied for every new ssh session.

MicPra, 2021-05-02:
Step by step... I made the alias permanent for all ssh sessions opened with the user 'tc' (in /home/tc/.profile). Opening a new shh session with PuTTY works fine! Even after reboot.

But with RasPi Check -  with the same login data as with PuTTY, I'll still need elevate 'tc' :-(

eidottermihi, 2021-05-03:
@MicPra, please enable debug logging in settings of the app and share your log here. Maybe there are some hints, why this is not working.

MicPra, 2021-05-03:
@eidottermihi, surley. Find the log attached. By the way - the location of vcgencmd within PiCore is:
./usr/local/bin/  (-> ./tmp/tcloop/rpi-vc/usr/local/sbin/)

I could try to set a link in ./usr/bin/

raspicheck.log:
`12:50:20.598 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
12:50:20.625 [AsyncTask #2] DEBUG d.e.rpicheck.activity.MainActivity - Device cursor rows: 4
12:50:20.763 [main] DEBUG d.e.rpicheck.activity.MainActivity - Spinner item selected: pos=0, id=1
12:50:24.904 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device)
12:50:24.904 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving new instance of all devices.
12:51:12.198 [main] INFO  d.e.r.activity.SettingsActivity - Disabled debug logging.
12:52:04.289 [main] WARN  d.e.r.activity.SettingsActivity - Enabling debug logging. Be warned that the log file can get huge because of this.
12:52:04.297 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
12:52:08.869 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
12:52:08.887 [AsyncTask #2] DEBUG d.e.rpicheck.activity.MainActivity - Device cursor rows: 4
12:52:09.020 [main] DEBUG d.e.rpicheck.activity.MainActivity - Spinner item selected: pos=0, id=1
12:52:15.259 [main] DEBUG d.e.rpicheck.activity.MainActivity - Spinner item selected: pos=2, id=4
12:52:15.270 [main] DEBUG d.e.rpicheck.activity.MainActivity - Switch from device id 1 to device id 4.
12:52:18.928 [main] DEBUG d.e.rpicheck.activity.MainActivity - Load average preference: FIVE_MINUTES
12:52:18.968 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - +++ Registered JCE providers +++
12:52:18.969 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: AndroidNSSP - Android Network Security Policy Provider
12:52:18.969 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: AndroidOpenSSL - Android's OpenSSL-backed security provider
12:52:18.969 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: CertPathProvider - Provider of CertPathBuilder and CertPathVerifier
12:52:18.969 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: AndroidKeyStoreBCWorkaround - Android KeyStore security provider to work around Bouncy Castle
12:52:18.969 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: HarmonyJSSE - Harmony JSSE Provider
12:52:18.969 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: AndroidKeyStore - Android KeyStore security provider
12:52:18.969 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: KnoxAndroidKeyStore - Knox Android KeyStore security provider
12:52:18.970 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: TimaKeyStore - TIMA KeyStore security provider
12:52:18.970 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Provider: SC - BouncyCastle Security Provider v1.56
12:52:18.982 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - +++ Availabe signatures +++
12:52:18.982 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA3-512WITHDDSA
12:52:18.982 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: WHIRLPOOLWITHRSA/ISO9796-2
12:52:18.982 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA3-224WITHRSAANDMGF1
12:52:18.982 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA384WITHDDSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA3-512WITHECDSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA3-256WITHECDDSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA1WITHRSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA512WITHRSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA3-512WITHRSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: NONEWITHRSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA256WITHDDSA
12:52:18.983 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA224WITHDDSA
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA512WITHECDSA
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: RMD128WITHRSA/X9.31
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: RSA
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA384WITHDSA
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA224WITHECDDSA
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA3-384WITHRSAANDMGF1
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA384WITHCVC-ECDSA
12:52:18.984 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: MD5WITHRSA
12:52:18.985 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA256WITHECNR
12:52:18.985 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: RIPEMD160WITHRSA/X9.31
12:52:18.985 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: SHA384WITHRSAANDMGF1
12:52:18.985 [AsyncTask #1] DEBUG d.e.rpicheck.ssh.impl.RaspiQuery - Signature: ECGOST3410
[... cut ...]
12:56:42.990 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
12:56:43.010 [AsyncTask #2] DEBUG d.e.rpicheck.activity.MainActivity - Device cursor rows: 4
12:56:43.132 [main] DEBUG d.e.rpicheck.activity.MainActivity - Spinner item selected: pos=0, id=1
12:56:46.520 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device)
12:56:46.521 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving new instance of all devices.
`

eidottermihi, 2021-05-03:
> Opening a new shh session with PuTTY works fine! Even after reboot.

RasPi Check just [executes a command](https://github.com/eidottermihi/rpicheck/blob/master/ssh/src/main/java/de/eidottermihi/rpicheck/ssh/impl/RaspiQuery.java#L281-L282) via SSH, PuTTy opens an interactive shell. Probably thats why aliasing has no effect, because `~/.profile` isnt evaluated in the first place.
But it was worth a try.

So right now I only see one solution to your problem: Try to convince PiCore folks to allow `vcgencmd` execution without sudo elevation (like all other Pi distros I know do it).
Implementing some workarounds in RasPi Check app is not a good way to go forward here, sorry.

I'm not closing the issue again, so maybe some other PiCore users find a proper solution or workaround on the side of PiCore and share it here.

MicPra, 2021-05-03:
Thank you very much for your investigation! Guess you are right - PiCore would be the place to solve. If there is a solution I get know from I will definitely post it here. Again - Thanks for the RasPi Check App!

eidottermihi, 2022-01-15:
closing now as its a PiCore issue, see https://github.com/eidottermihi/rpicheck/issues/211#issuecomment-831526798

closed 2022-01-15 by eidottermihi