eidottermihi/rpicheck #223: OSMC Pi gets "A transport exception occurred"

TulsaTV, 2021-12-21:
I have RasPi Check working on three different tablets. I successfully check my Pi1 and Pi 3 devices running Pi OS, and two Pi 3s running OSMC. The widgets work.

Got a new Samsung S20 FE 5G Android phone. RasPi Check works there as well for the two Pi OS boxes and one of the OSMC boxes. The other OSMC box returns "A transport exception occurred".

SSH works fine on the latter box. I can do df and ps commands, no problem. On one single occasion, RasPi Check did work on this box, but no other time.

What could possibly be the difference? The two boxes are set up identically, other than the differing fixed IP addresses.

TulsaTV, 2022-01-13:
As of the Samsung S20 FE 5G upgrade to Android 12 today, RasPi Check on that device no longer works on any of my Pis. Still works fine on other Android devices.

mcmellow, 2022-04-13:
On  Poco X3 Pro with android 12 (last update 2022-02-01)  got the same error.  **Transport exception occurred**
The app worke fine until some weeks ago, cannot use the app at all.  What can I do?

rpvelloso, 2022-04-15:
Same error on android 12 moto edge 20

vitrudchenko, 2022-07-29:
Any news on it?

mcmellow, 2022-08-28:
Din not worked out OK. Removed the app from my smart phone