eidottermihi/rpicheck #226: Connection with Bullseye ?

Trotter73, 2022-02-03:
Hi,
I seem to be having issues connecting to Pis running Bullseye, I get "Oops, could not establish a connection. Please check hostname and your network. "

I'm connecting with a pub key and passphrase, as a control; fresh install of Buster & Bullseye, just updates applied, same keys installed on both, both cards booted on the same PiZeroW. I am able to connect to both using my normal SSH client (Bitvise).

This is the debug log for a failed connection.

I am using an RSA key...

```
13:09:52.755 [main] WARN  d.e.r.activity.SettingsActivity - Enabling debug logging. Be warned that the log file can get huge because of this.
13:09:52.769 [main] DEBUG d.e.r.activity.helper.LoggingHelper - Logging was configured, debug logging enabled: true
13:10:00.592 [main] DEBUG d.e.rpicheck.activity.MainActivity - Load average preference: FIVE_MINUTES
13:10:00.594 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - New RaspiQuery for host: 192.168.1.123
13:10:00.596 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Connecting to host: 192.168.1.123 on port 1234.
13:10:00.624 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using no host key verification.
13:10:00.624 [AsyncTask #1] INFO  d.e.rpicheck.ssh.impl.RaspiQuery - Using private/public key authentification with passphrase.
13:10:01.501 [reader] ERROR n.s.sshj.transport.TransportImpl - Dying because - {}
net.schmizz.sshj.transport.TransportException: Unable to reach a settlement: [diffie-hellman-group14-sha1, diffie-hellman-group1-sha1] and [curve25519-sha256, curve25519-sha256@libssh.org, ecdh-sha2-nistp256, ecdh-sha2-nistp384, ecdh-sha2-nistp521, diffie-hellman-group-exchange-sha256, diffie-hellman-group16-sha512, diffie-hellman-group18-sha512, diffie-hellman-group14-sha256]
at net.schmizz.sshj.transport.Proposal.firstMatch(Proposal.java:145) ~[na:0.0]
at net.schmizz.sshj.transport.Proposal.negotiate(Proposal.java:127) ~[na:0.0]
at net.schmizz.sshj.transport.KeyExchanger.gotKexInit(KeyExchanger.java:219) ~[na:0.0]
at net.schmizz.sshj.transport.KeyExchanger.handle(KeyExchanger.java:344) ~[na:0.0]
at net.schmizz.sshj.transport.TransportImpl.handle(TransportImpl.java:484) ~[na:0.0]
at net.schmizz.sshj.transport.Decoder.decode(Decoder.java:107) ~[na:0.0]
at net.schmizz.sshj.transport.Decoder.received(Decoder.java:175) ~[na:0.0]
at net.schmizz.sshj.transport.Reader.run(Reader.java:61) ~[na:0.0]
13:10:01.504 [AsyncTask #1] ERROR net.schmizz.concurrent.Promise - <<kex done>> woke to: net.schmizz.sshj.transport.TransportException: Unable to reach a settlement: [diffie-hellman-group14-sha1, diffie-hellman-group1-sha1] and [curve25519-sha256, curve25519-sha256@libssh.org, ecdh-sha2-nistp256, ecdh-sha2-nistp384, ecdh-sha2-nistp521, diffie-hellman-group-exchange-sha256, diffie-hellman-group16-sha512, diffie-hellman-group18-sha512, diffie-hellman-group14-sha256]
13:10:01.514 [AsyncTask #1] ERROR d.e.rpicheck.activity.SSHQueryTask - Could not establish a connection to host '192.168.1.123:1234'. Check your connection and settings.
de.eidottermihi.rpicheck.ssh.impl.RaspiQueryException: Could not establish a connection to host '192.168.1.123:1234'. Check your connection and settings.
at de.eidottermihi.rpicheck.ssh.impl.RaspiQuery.connectWithPubKeyAuthAndPassphrase(RaspiQuery.java:1140) ~[na:0.0]
at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:92) ~[na:0.0]
at de.eidottermihi.rpicheck.activity.SSHQueryTask.doInBackground(SSHQueryTask.java:46) ~[na:0.0]
at android.os.AsyncTask$3.call(AsyncTask.java:378) ~[na:0.0]
at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[na:0.0]
at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:289) ~[na:0.0]
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) ~[na:0.0]
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) ~[na:0.0]
at java.lang.Thread.run(Thread.java:919) ~[na:0.0]
Caused by: net.schmizz.sshj.transport.TransportException: Unable to reach a settlement: [diffie-hellman-group14-sha1, diffie-hellman-group1-sha1] and [curve25519-sha256, curve25519-sha256@libssh.org, ecdh-sha2-nistp256, ecdh-sha2-nistp384, ecdh-sha2-nistp521, diffie-hellman-group-exchange-sha256, diffie-hellman-group16-sha512, diffie-hellman-group18-sha512, diffie-hellman-group14-sha256]
at net.schmizz.sshj.transport.Proposal.firstMatch(Proposal.java:145) ~[na:0.0]
at net.schmizz.sshj.transport.Proposal.negotiate(Proposal.java:127) ~[na:0.0]
at net.schmizz.sshj.transport.KeyExchanger.gotKexInit(KeyExchanger.java:219) ~[na:0.0]
at net.schmizz.sshj.transport.KeyExchanger.handle(KeyExchanger.java:344) ~[na:0.0]
at net.schmizz.sshj.transport.TransportImpl.handle(TransportImpl.java:484) ~[na:0.0]
at net.schmizz.sshj.transport.Decoder.decode(Decoder.java:107) ~[na:0.0]
at net.schmizz.sshj.transport.Decoder.received(Decoder.java:175) ~[na:0.0]
at net.schmizz.sshj.transport.Reader.run(Reader.java:61) ~[na:0.0]
13:10:01.516 [AsyncTask #1] DEBUG d.e.rpicheck.activity.SSHQueryTask - Query time: 921 ms.
13:10:01.526 [main] DEBUG d.e.rpicheck.activity.MainActivity - Query caused exception. Showing dialog.
13:10:09.258 [main] DEBUG d.e.rpicheck.activity.MainActivity - Saving instance state (current device)
13:10:09.259 [main] DEBUG d.e.rpicheck.activity.MainActivity - Adding current device to all
```

Trotter73, 2022-03-13:
Hi,

I think I have found the issue, in Debian 10  the KexAlgorithms diffie-hellman-group14-sha1 has been depreciated because of security concerns, it appears rpicheck only uses this and so connections fail on 10..

You can get around this by editing /etc/ssh/sshd_config  and adding KexAlgorithms diffie-hellman-group14-sha1  I cant recommend though because of the security concern.

jsmr1, 2022-08-11:
Yes thats helps. Don't forget the "+" in order to add this method to the defaults instead of replacing all methods 

 KexAlgorithms +diffie-hellman-group14-sha1